nathankellenicki/node-poweredup #1: Pup remote voltage sensor

cpseager, 2018-08-13:
Upon powerup the pup hub reports the following internal devices as new port connections

     [15, 0, 4, 50, 1, 23, 0, 0, 0, 0, 16, 0, 0, 0, 16]
     [15, 0, 4, 59, 1, 21, 0, 2, 0, 0, 0, 2, 0, 0, 0]
     [15, 0, 4, 60, 1, 20, 0, 2, 0, 0, 0, 2, 0, 0, 0]

port 0x32 device 0x17 (rgb led)
port 0x3B device 0x15 (current sensor)
port 0x3C device 0x14 (voltage sensor)

However the pup remote reports 

     [15, 0, 4, 0, 1, 55, 0, 0, 0, 0, 16, 0, 0, 0, 16]
     [15, 0, 4, 1, 1, 55, 0, 0, 0, 0, 16, 0, 0, 0, 16]
     [15, 0, 4, 52, 1, 23, 0, 0, 0, 0, 16, 0, 0, 0, 16]
     [15, 0, 4, 59, 1, 20, 0, 2, 0, 0, 0, 2, 0, 0, 0]
     [15, 0, 4, 60, 1, 56, 0, 0, 0, 0, 16, 0, 0, 0, 16]

port 0x00 device 0x37 (remote buttons)
port 0x01 device 0x37 (remote buttons)
port 0x34 device 0x17 (rgb LED)
port 0x3B device 0x14 (voltage sensor)
port 0x3C device 0x38 (unknown different sensor)

Note the 0x14 voltage sensor is on different ports.

Therefore assuming internal port 0x3C is always going to be voltage in _parseSensorMessage  is not correct. The internal sensors should really be programmatically 'attached' to their port in exactly the same way that external devices are attached to a port - the necessary 'new device' notification is transmitted by the hub when the initial connection is made.

nathankellenicki, 2018-08-13:
Thanks for that! I hadn't noticed that the voltage sensor had swapped from 0x3c to 0x3b on the PUP remote. I'll fix this.

Im aware that the port notifications for these devices are issued as normal, but my current implementation exposes ports as user facing, and I'd rather not expose these virtual ports for devices that can't attach/detach physically. I consider it a bit user unfriendly to expose the implementation detail of that. :) 

I am however considering revamping the code to expose peripheral objects on a port, and assigning the control methods (setMotorSpeed etc) to the peripheral object rather than the port, but we'll see...

nathankellenicki, 2018-08-13:
I also hadn't noticed 0x38. Do you have any idea what it does? Interesting that the buttons are 0x37 and this is 0x38, so perhaps something related to the buttons/some other hidden remote functionality...

cpseager, 2018-08-13:
Yes, is  anew sensor type, I was hoping for tilt but the PCB doesn't appear to have the correct chip! I suspect it may just be a 'different type' of current sensor purely because the value changes slightly but not much- its a guess but 16 bit value / 1000 gives an acceptable value in mA.

nathankellenicki, 2018-08-14:
Gotcha. It does appear to look like a reasonable value, so I'll use it for current for now. :) Thanks