nathankellenicki/node-poweredup #7: [Question] Rotation - how to interpret the numbers?

draGy-, 2018-12-24:
Hey there,

I was confused about the information the "rotation angle of Motor x" in the Boost-App provided, so I tried to check out, whether I could besser Information using this API.

But do be honest, I am still puzzled. Could you provide more insight about the number that the "rotation" event it emitting?

I was expecting a number in degrees, but instead I get really high numbers and after rotating the motor I do not get the same numbers again, although the motor/tacho was positioned back to the original position.

I hope you can help me out.

Thanks.

nathankellenicki, 2018-12-25:
Hi, and Merry Christmas!

The numbers can get large very quickly as iirc a full rotation is 360 (so two is 720 etc), and can take a split second. If you’re watching the numbers while you’ve got a speed set they’ll get large VERY quickly.

Which motor, hub, and functions are you using?

draGy-, 2018-12-25:
Hey (and also merry Christmas to you), 

I am using the move Hub from the Lego Boost set.

This is my test code:
```
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();
poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    await hub.connect(); // Connect to the Hub
    await hub.sleep(3000); // Sleep for 3 seconds before starting

    hub.on("rotate", async (port, rotation) => {
        if (port == 'A' || port == 'B') {
            console.log("rotate(" + port + ", " + rotation + ")");
        }
    });
});

poweredUP.scan(); // Start scanning for Hubs
```

And I am trying to use one of the integrated motors/tachos as a lever. So I manually move the lever a bit (less than 90 degrees).
But even the smallest movement gives me numbers I do not understand:

Values:
```
rotate(A, -1558715)
rotate(A, -1951931)
rotate(A, -2672827)
rotate(A, -2934971)
rotate(A, -3000507)
rotate(A, -3459259)
rotate(A, -3721403)
rotate(A, -3786939)
rotate(A, -3852475)
rotate(A, -4376763)
rotate(A, -4573371)
rotate(A, -4638907)
rotate(A, -4835515)
rotate(A, -4901051)
rotate(A, -4835515)
rotate(A, -4442299)
rotate(A, -4245691)
rotate(A, -4180155)
rotate(A, -4049083)
rotate(A, -3590331)
rotate(A, -3524795)
rotate(A, -3197115)
rotate(A, -3066043)
rotate(A, -3000507)
rotate(A, -2803899)
rotate(A, -2738363)
rotate(A, -2672827)
rotate(A, -2476219)
rotate(A, -2410683)
```

nathankellenicki, 2018-12-26:
Interesting - admittedly it's been a while since I tested this code, something might have changed.

I have no Lego on my over Christmas - are you able to run a quick experiment for me?

Could you rotate a motor by approx 90 degrees and give me the number? And do the same 4 times? Basically paste the values for four seperate quarter rotations.

I'll try and work out the numbers from that.

draGy-, 2018-12-26:
Values, when I rotate manually
First 90 degrees:
```
rotate(A, 14149)
rotate(A, -116923)
rotate(A, -1755323)
rotate(A, -3393723)
rotate(A, -4180155)
rotate(A, -5163195)
rotate(A, -5556411)
```

next 90 degrees:
```
rotate(A, -5621947)
rotate(A, -7129275)
rotate(A, -8833211)
rotate(A, -9488571)
rotate(A, -10078395)
rotate(A, -10733755)
rotate(A, -11061435)
rotate(A, -11323579)
```

next 90 degrees:
```
rotate(A, -11389115)
rotate(A, -11585723)
rotate(A, -13289659)
rotate(A, -14469307)
rotate(A, -15583419)
rotate(A, -16697531)
rotate(A, -17352891)
rotate(A, -17746107)
rotate(A, -17877179)
```

next 90 degrees:
```
rotate(A, -17942715)
rotate(A, -18335931)
rotate(A, -20105403)
rotate(A, -21940411)
rotate(A, -22661307)
rotate(A, -22726843)
rotate(A, -22792379)
rotate(A, -22857915)
rotate(A, -23513275)
rotate(A, -23578811)
```

This are the valuess when I perform the following code

```
hub.on("rotate", async (port, rotation) => {
        if (port == 'A' || port == 'B') {
            console.log("rotate(" + port + ", " + rotation + ")");
        }
    });

    
    console.log("move motor for 90 degrees - 1");
    await hub.setMotorAngle("A", 90);

    console.log("move motor for 90 degrees - 2");
    await hub.setMotorAngle("A", 90);

    console.log("move motor for 90 degrees - 3");
    await hub.setMotorAngle("A", 90);

    console.log("move motor for 90 degrees - 4");
    await hub.setMotorAngle("A", 90);
```

```
move motor for 90 degrees - 1
rotate(A, -23513275)
rotate(A, -22988987)
rotate(A, -21219515)
rotate(A, -18729147)
rotate(A, -17483963)
move motor for 90 degrees - 2
rotate(A, -17418427)
rotate(A, -17352891)
rotate(A, -15780027)
rotate(A, -13420731)
rotate(A, -11847867)
rotate(A, -11716795)
move motor for 90 degrees - 3
rotate(A, -11651259)
rotate(A, -11454651)
rotate(A, -9357499)
rotate(A, -6670523)
rotate(A, -5621947)
rotate(A, -5753019)
move motor for 90 degrees - 4
rotate(A, -5687483)
rotate(A, -5621947)
rotate(A, -4901051)
rotate(A, -2934971)
rotate(A, -575675)
rotate(A, 210757)
```

nathankellenicki, 2018-12-26:
Aha, thank you kindly. I *think* I've fixed the issue, I was reading the int32 from byte position 2 rather than 4. Of course I can't test, but I'm pretty sure this is the issue.

Could you try with latest version of this package (v1.2.0)?

draGy-, 2018-12-27:
Hi Nathan,

great the fix worked,

Here are the results:
```

move motor for 90 degrees - 1
rotate(A, 0)
rotate(A, 11)
rotate(A, 41)
rotate(A, 79)
rotate(A, 91)
move motor for 90 degrees - 2
rotate(A, 92)
rotate(A, 95)
rotate(A, 122)
rotate(A, 161)
rotate(A, 182)
rotate(A, 183)
move motor for 90 degrees - 3
rotate(A, 184)
rotate(A, 202)
rotate(A, 239)
rotate(A, 269)
rotate(A, 271)
move motor for 90 degrees - 4
rotate(A, 272)
rotate(A, 285)
rotate(A, 318)
rotate(A, 354)
rotate(A, 362)
rotate(A, 363)
```

closed 2018-12-27 by draGy-