nathankellenicki/node-poweredup #10: scanning start and stop

firien, 2019-01-09:
I am tinkering with a [web app](https://github.com/firien/duplo-train) to control some Duplo trains. I want it to handle multiple trains and be robust enough to deal with them powering on and off. (My thought was to have this running on a Pi Zero 24/7)

The scan function for `PoweredUP` stops scanning after a discovery. Calling `scan` again seems to add duplicate event listeners for the Noble `discover` event. Was it not designed for the scan to start and stop?

* Is there a reason you stop scanning ([???](https://github.com/noble/noble#adapter-specific-known-issues))
* Would you accept a PR to handle start/stops?

----

thanks for making this project

nathankellenicki, 2019-01-09:
I'm afraid I experience different behaviour to you; in fact, I've specifically included a workaround to get around common reasons why it stops scanning after a discovery - see here: https://github.com/nathankellenicki/node-poweredup/blob/master/src/poweredup.ts#L87-L89

I myself connect to multiple hubs in multiple Lego trains with multiple remotes. I also connect boost hubs with PUP remotes.

So, to answer your questions:

1. I don't. :)
2. Yes of course, assuming we discover why it stops scanning for you in the first place.

So, can you provide a little more information? What OS/Node version/BLE adapter/etc?

firien, 2019-01-09:
I don't actually have multiple hubs to connect to yet - so I didn't experience it - just thought I noticed a bug while reading the code. But I definitely missed the bang! when i read it - thought it only started again _for_ browser contexts.

nathankellenicki, 2019-01-09:
Ah I see. Yes, for browser contexts it explicitly doesn't call scan again. The reason for this is that doing so with the web-bluetooth spec displays a modal dialog box asking the user to select the Bluetooth device they want. Therefore repeated calls to scan is the most desirable behaviour as they would have to choose their devices individually.

However the browser context functionality isn't fully implemented in this library at the moment for various reasons.

Let me know if you experience discovery issues when you have multiple hubs to test with. In the meantime, I'll work on improving the scan event listener attachments.

nathankellenicki, 2019-01-09:
Fixed in #363eb5f

closed 2019-01-09 by nathankellenicki

nathankellenicki, 2019-01-09:
Fixed with 363eb5f4d0f4e04f623c6c6f22a1322fa1e54dde

commit 363eb5f4 (2019-01-09): Fixed bug where calling scan multiple times adds multiple discovery events, closes #10