nathankellenicki/node-poweredup #11: Getting hub object if you know hub uuid

MihaZelnik, 2019-01-10:
Is it possible to get hub object if you know hub uuid? 

So you do something like that, where you can get hubUuid from params for example?
```
hub = PUPHub(hubUuid);
hub.setLEDColor(1);
```

nathankellenicki, 2019-01-10:
This should do what you want: https://github.com/nathankellenicki/node-poweredup#PoweredUP+getConnectedHubByUUID

MihaZelnik, 2019-01-10:
Ah nice, totally missed that one :/

MihaZelnik, 2019-01-10:
BTW awesome lib!

closed 2019-01-10 by MihaZelnik

nathankellenicki, 2019-01-10:
No problem, and thanks! :)

MihaZelnik, 2019-01-11:
@nathankellenicki I did use your package and created REST API on top of it https://github.com/MihaZelnik/PoweredUp-REST-API/ (this is just a quick prototype for now so I know that it is possible)

Idea would be to have raspberry pi as a server so I can communicate with hubs via http and create some kind of web dashboard where I will be able to control it and make scenarios

GianCann, 2019-02-01:
Hi, @nathankellenicki 
can you add a new method called getConnectedHubByName, that use friendly name configured by setName, instead of UUID?