nathankellenicki/node-poweredup #13: Getting error [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to write outside buffer bounds

cristian-spiescu, 2019-01-20:
Hello,

I just discovered today this very interesting project, but I'm having some trouble running it. I'm using Windows 10 and yesterday I managed to run successfully the project [movehub-async](https://github.com/ttu/node-movehub-async) . `movehub-async` has a narrower scope than this project; i.e. it works only with Lego Boost, whereas this project, `node-poweredup`, seems to support all devices using the [Lego Wireless Protocol (LWP)](https://lego.github.io/lego-ble-wireless-protocol-docs). I mention that I installed [all noble prerequisites](https://github.com/noble/noble#prerequisites), including installing the WinUSB driver.

The first issue I got with both projects was:
```
Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (D:\temp\test-js-ble2\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (D:\temp\test-js-ble2\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (D:\temp\test-js-ble2\node_modules\noble-mac\node_modules\noble\lib\hci-socket\hci.js:101:35)
    at NobleBindings.init (D:\temp\test-js-ble2\node_modules\noble-mac\node_modules\noble\lib\hci-socket\bindings.js:82:13)
    at Noble.<anonymous> (D:\temp\test-js-ble2\node_modules\noble-mac\node_modules\noble\lib\noble.js:57:24)
```

Having an Intel Bluetooth module, I tried my luck (as some [google results suggested](https://github.com/noble/noble/issues/472)) and I modified the file `\node_modules\bluetooth-hci-socket\lib\usb.js`; i.e. adding the signature of my bluetooth device, as seen in Device Manager. Something like:

```
    this._usbDevice = ... || usb.findByIds(0x8087, 0x0aaa);
```

After doing this, project `node-movehub-async` has ran successfully. However, in this project, the error was gone, but I get the following (cf. the title of this issue).

```
RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to write outside buffer bounds
    at boundsError (internal/buffer.js:51:11)
    at Buffer.readUInt8 (internal/buffer.js:141:5)
    at Hci.processCmdCompleteEvent (D:\temp\test-js-ble2\node_modules\noble-mac\node_modules\noble\lib\hci-socket\hci.js:566:25)
    at Hci.onSocketData (D:\temp\test-js-ble2\node_modules\noble-mac\node_modules\noble\lib\hci-socket\hci.js:461:12)
    at BluetoothHciSocket.emit (events.js:182:13)
    at BluetoothHciSocket.onHciEventEndpointData (D:\temp\test-js-ble2\node_modules\bluetooth-hci-socket\lib\usb.js:156:10)
    at InEndpoint.emit (events.js:182:13)
    at Transfer.transferDone (D:\temp\test-js-ble2\node_modules\usb\usb.js:441:9)
```
Have you got any hints? I'm kind of stuck.

Note: just modifying the `usb.js` file cf. above was theory is an illegal hack. I.e. I bypass the guard condition, but maybe then the lib may fail somewhere else, because my Bluetooth device is different than the ones already supported. E.g. with errors such as the one above. BUT, given the fact that it has actually worked for the other project, `node-movehub-async`, makes me hope that maybe there might be a solution.

Thanks in advance!
Cristian

nathankellenicki, 2019-01-20:
Hello! I'm afraid I don't use Windows 10, have access to a Windows 10 machine, nor know anyone who uses Windows 10 for that matter, so this may be a little tricky to debug!

But for starters, can you give me a little background. What version of Node are you? What type of Hub are you trying to connect to, and what devices (such as motors or sensors) are plugged into it?

Also, can you tell me at what point you start getting these errors? Immediately after requiring the library? After discovering a Hub? After calling connect on the hub? Or after attempting to call a method (such as setMotorSpeed) on a discovered and connected Hub?

nathankellenicki, 2019-01-30:
Hi @cristian-spiescu, do you have a further update on this?

nathankellenicki, 2019-02-25:
Closing due to lack of activity, info, and corroborating reports.

closed 2019-02-25 by nathankellenicki