nathankellenicki/node-poweredup #14: subscribe/unsubscribe usage

MihaZelnik, 2019-01-24:
I'm trying to figure it out what subscribe/unsubscribe functions are doing?

For color sensor I'm using hub.on("color"), could I use subscribe/unsubscribe for color sensor, or are this functions used for some other use-case/device

Also just saw that you will implement power off https://www.eurobricks.com/forum/index.php?/forums/topic/162288-powered-up-a-tear-down/&page=13&tab=comments#comment-3068973 I was just thinking how could I do that, since if you disconnect you are automatically connected back

nathankellenicki, 2019-01-24:
The subscribe/unsubscribe methods are used at connection time of a new device (color sensor, tilt sensor, etc) to tell the sensor what mode to put itself in. It is automatically called by this library.

For example, the "color and distance sensor" shipped with the Boost tookit has several modes - one where it only reports color values, one where it only reports distance values, one where it reports both, etc. There are a couple more.

The reason they exist sn that by only reporting the values the receiver cares about, it can save BLE bandwidth by not sending out the rest of the data that the receiver has indicated it doesn't care about.

For simplicity's sake, this library automatically puts the sensor into "color and distance" mode, as you can just ignore the values you don't care about, and I've yet to run into any issues (or hear of anyone running into issues) with BLE bandwidth limits.

If you want (and the reason these methods are public rather than private), you can set the sensors into different modes - but, WARNING! :) I have not tested the sensor message parsing code with other modes other than the defaults I've set, and as they currently don't check, it *may* break!

Regarding your second question - you will only reconnect if you are still scanning. So you should call stop() on your PoweredUP instance before you disconnect your Hub instance.

However I believe the power off talked about in that thread is a hard power off - immediate shutdown, no blinking lights, no attempt to reconnect. Almost like flicking a switch. But I haven't tested it yet.

MihaZelnik, 2019-01-24:
thanks, now it makes more sense :) I had a feeling that it is something more low level

nathankellenicki, 2019-01-24:
Well that would only happen if a) both devices were disconnected at the same time, and b) both the remote and the hub had been paired together at some point in the past.

But agreed, a hard shutdown would be good (if only to stop other devices/people who are scanning from connecting to it), hopefully it works as advertised. :)

nathankellenicki, 2019-01-24:
Aaannd it does. A single blink and the device goes off. No attempt to connect to anything else. :)

I'll look to publishing a new version with this functionality a bit later today.

closed 2019-01-25 by MihaZelnik