nathankellenicki/node-poweredup #15: Motor don't run...

GianCann, 2019-01-30:
I have one Smart Hub and one train motor from LEGO City Train.
I haved used Raspberry 3 B+ with last Raspbian version.
After installed Noble Library (https://github.com/noble/noble), and Node-Poweredup, i have activate the BT and execute the example to start/stop motor for 2 second with this command:
node lego.js

The Led on Smart Hub becomes fixed, but the motor (connected to port A) don't start.

The console report:
noble warning: unknown peripheral 90842b00420c
noble warning: unknown peripheral 90842b00420c
noble: unknown peripheral null connected!
noble: unknown peripheral null connected!
noble: unknown peripheral null connected!

What am I doing wrong?

nathankellenicki, 2019-01-30:
Could you tell me which version you have (in the package.json)? And then could you set "export DEBUG=poweredup,lpf2hub,puphub" in your console before you run node, and send me some more debug from that?

Thanks!

GianCann, 2019-01-30:
(I've first restart the Raspberry)

Now i see:

**pi@raspberrypi:~ $ export DEBUG=poweredup,lpf2hub,puphub
pi@raspberrypi:~ $ node lego.js**
  puphub Discovered Powered UP Hub +0ms
  puphub Discovered Powered UP Hub +28ms
  puphub Discovered Powered UP Hub +31ms
  puphub Discovered Powered UP Hub +28ms
  puphub Discovered Powered UP Hub +98ms
  puphub Discovered Powered UP Hub +34ms
  puphub Discovered Powered UP Hub +30ms
  puphub Discovered Powered UP Hub +29ms
  puphub Discovered Powered UP Hub +26ms
  puphub Discovered Powered UP Hub +30ms
  puphub Discovered Powered UP Hub +30ms
  puphub Discovered Powered UP Hub +61ms
  puphub Discovered Powered UP Hub +31ms
  puphub Connecting to Powered UP Hub +46ms
  puphub Connecting to Powered UP Hub +134ms
  puphub Connecting to Powered UP Hub +1ms
  puphub Connecting to Powered UP Hub +0ms
  puphub Connecting to Powered UP Hub +48ms
  puphub Connecting to Powered UP Hub +35ms
  puphub Connecting to Powered UP Hub +30ms
  puphub Connecting to Powered UP Hub +28ms
  puphub Connecting to Powered UP Hub +26ms
  puphub Connecting to Powered UP Hub +31ms
  puphub Connecting to Powered UP Hub +30ms
  puphub Connecting to Powered UP Hub +60ms
  puphub Connecting to Powered UP Hub +31ms
(node:1011) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 disconnect listeners added. Use emitter.setMaxListeners() to increase limit
(node:1011) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 servicesDiscover listeners added. Use emitter.setMaxListeners() to increase limit

But the motor always don't run

nathankellenicki, 2019-01-30:
Thanks. I think I see the problem, and it's related to a change I made just yesterday. Could you try v1.7.4?

I think the problem might be related to a delay I added to discovery to wait for all Bluetooth advertisement packets to arrive. Introducing this delay added asynchronicity to the discovery process. This conflicts with a restart of the noble discovery process I have to get round some discovery issues on some platforms (Which I might have to fix some other way!)

GianCann, 2019-01-30:
> Thanks. I think I see the problem, and it's related to a change I made just yesterday. Could you try v1.7.4?

Please explain how to do this (I'm new on node.js)

nathankellenicki, 2019-01-30:
How did you install the original version you have? :) Either:

a) git clone this repo again (which will get you latest)
b) Delete your node_modules directory and re-run npm install node-poweredup (which will also get you latest)

See here for more info: https://docs.npmjs.com/downloading-and-installing-packages-locally

nathankellenicki, 2019-01-30:
I think the easiest route for you to get started with this project is:

1. Create a new directory to put your project in, and change to that project.
2. Run "npm init -y" to initialize the project with a package.json file.
3. Run "npm install node-poweredup --save" which will pull this library into your project's node_modules directory, and save the version to your package.json.
4. You can now write code in that directory that uses node-poweredup.

GianCann, 2019-01-30:
Ok, now it's work well!.
Thank you.

Can you tell how can integrate this node in Node Red to send command to start/stop/change speed?

nathankellenicki, 2019-01-30:
I'm afraid I don't use Node red, however from a cursory look at the documentation, you want to follow these guides for creating a "node" which you can use in your Node red pipeline - https://nodered.org/docs/creating-nodes/

nathankellenicki, 2019-01-30:
Closed, resolved with 748c886df5ee5ea3ca97138248a3b6f58f6ebba1