nathankellenicki/node-poweredup #16: Feature request: getConnectedHubByName

nathankellenicki, 2019-02-01:
Hi, @nathankellenicki 
can you add a new method called getConnectedHubByName, that use friendly name configured by setName, instead of UUID?

_Originally posted by @GianCann in https://github.com/nathankellenicki/node-poweredup/issues/11#issuecomment-459745886_

nathankellenicki, 2019-02-01:
Hi @GianCann, I've added this in v1.8.0 which is now available. It is *almost* as you asked - getConnectedHubsByName (notice the plural), which returns an array of matching hubs. This is because multiple connected hubs can have the same name.

Hopefully this does what you need. :)

GianCann, 2019-02-01:
ok, thank you very much!

nathankellenicki, 2019-02-01:
Closed, fixed with 9ced6da24e2f45536d0253429542d73047a69efb

closed 2019-02-01 by nathankellenicki