nathankellenicki/node-poweredup #18: setLEDColor strange behavior and connection method

GianCann, 2019-02-05:
I running your library on Raspberry 3 B+ (Raspbian os).
I have 4 Poweredup Hub, with 4 train motors connected to port A of each hub.

I put to power on all 4 hubs and start scanning with this portion of code:

```
let red_train = null;
let blue_train = null;
let green_train = null;
let yellow_train = null;
let iHubCount=0;

poweredUP.scan();
console.log("Start scan...");

poweredUP.on("discover", async (hub) => {
    await hub.connect();
    
    if (iHubCount==4){
        yellow_train=hub;
        yellow_train.sleep(500);
        yellow_train.setLEDColor(COLORS["yellow"]);
        iHubCount=5;
        //poweredUP.scan();
    }
    
    if (iHubCount==3){
        green_train=hub;
        green_train.sleep(500);
        green_train.setLEDColor(COLORS["green"]);
        iHubCount=4;
        //scan for new hub
        //poweredUP.scan();
    }
    
    if (iHubCount==1){
        blue_train=hub;
        blue_train.sleep(500);
        blue_train.setLEDColor(COLORS["blue"]);
        iHubCount=2;
        //scan for new hub
        //poweredUP.scan();
    }
    
    if (iHubCount==0){
        red_train=hub;
        red_train.sleep(500);
        red_train.setLEDColor(COLORS["red"]);
        iHubCount=1;
        //scan for new hub
        //poweredUP.scan();
    }
    
    //if (hub.name=="Gianluca"){
    //    hub.setLEDColor(9);
    //}
    
    console.log(`Connected to ${hub.name} (${hub.uuid})`);
    sLog += `Connected to ${hub.name} (${hub.uuid})\n`;

    //hub.on("disconnect", () => {
    //    console.log(`Hub ${hub.name} (${hub.uuid}) disconnected`);
    //    sLog += `Hub ${hub.name} (${hub.uuid}) disconnected\n`;
    //})
});
```
I want to assign a different color to each hub, in order of 'connect' event.
For first two hubs, the color are set right. For others hubs, remaining white.
If i set color, later, from another function, the LED color change correctly.
What i wrong?

Another question: dhat do you think is the best way to look forward to new hubs without first having to power on (all together) and then launch the scan?
I want to start scanning when Raspberry boot up and rest in waiting that 1, 2, 3 or 4 hubs, powered in different moments.

If you look the portion of the code i have try to call poweredUP.scan() after each hub connection.
it work, but sometime don't find the new hubs powered

firien, 2019-02-05:
`iHubCount` is never set to 3. I assume you are getting red and blue, but not green and yellow.

GianCann, 2019-02-06:
Damn't!!! I'm very stupid :(
Thank you @firien

nathankellenicki, 2019-02-08:
Great, thank you both, and glad you got it sorted! :)

closed 2019-02-08 by nathankellenicki