nathankellenicki/node-poweredup #19: Boost Hub and setMotorAngle gives error but works

tregusti, 2019-02-10:
Hi,

first of all, I was really happy when I found this library you made!

I have a LEGO Boost Hub and want to rotate the hub motor by a certain degree. So I try `setMotorAngle` but I get this error: `Angle rotation is only available when using a Boost Tacho Motor or Boost Move Hub Motor`. This is due a if statement checking the `type` on the `portObj` used. However, there is only one line in the whole code base settning that property on a port, in the [contructor](https://github.com/nathankellenicki/node-poweredup/blob/2e68ae54094c62eaa71bd403bdb697c2a8fc9883/src/port.ts#L16-L20) of `Port`. And there it's set to `UNKNOWN`. So the `if` will never be fulfilled.

If I just remove the `if` regarding `portObj.type` in the `setMotorAngle` method in `BoostMoveHub` all works as intended.

What is the idea here, that the port type should perhaps be added to the port constructor? Or should it be set later by the hub object? Or should the `if` in fact be removed? Or…

Using `1.9.1`.

Thank you!

nathankellenicki, 2019-02-10:
Hi, that's very strange!

The actual line that sets the type of the portObj is [here](https://github.com/nathankellenicki/node-poweredup/blob/master/src/hub.ts#L329). It's set when we receive a message from the hub telling us that its identified the device on a port.

I thought this might be a regression but I've just tested on one of my Boost Move Hubs and it Works On My Machine™, so I wonder if it's that you're calling `setMotorAngle` before the hub has correctly identified the device and told us about it.

Can you try either waiting for a couple of seconds before calling the methods (either setTimeout or using the hub.sleep helper method), or even better, watching out for the "attach" event the hub emits? For example:

```javascript
hub.on("attach", (port, device) => {
    if (port === "D") {
        hub.setMotorAngle("D", 170, 100);
    }
});

hub.connect();
```

tregusti, 2019-02-10:
Sorry, my search fu is handicapped it seems.

I was using the example code from the readme. And instead or doing the `while (true)` loop, I just invoked `setMotorAngle` instead.

I will try the things you mentioned. But If that is needed. Some more text would be good to explain the needs in the docs =)

nathankellenicki, 2019-02-10:
Yeah, the sample code is simple to distill the idea of the library rather than contain all the checks needed to be error free, but perhaps that's not the best approach. It has a 3 second delay which *should* be enough time for us to have received the necessary messages from the hub, but I suppose Bluetooth distance, interference, etc doesn't make this guaranteed.

I shall have a think about revamping the sample code. In the meantime, let me know if this works. :)

tregusti, 2019-02-10:
Waiting 3 seconds solved my issue. But the motors in the Boost Hub itself should be known on connect I assume?

Thank you for quick response!

nathankellenicki, 2019-02-10:
Unfortunately not! Despite being physically part of the hub, the hub firmware treats the built in motors no different from any other attached device, and will wait to identify them before sending out the device identification message and triggering the attach event on the relevant ports.

This is a consistent theme across all LPF2 devices. For example, the Powered UP remote (Yes, it is also a "hub") will send out messages saying that it has identified the buttons on the face of the remote, despite them being physically built in!

nathankellenicki, 2019-02-11:
Closing as resolved

closed 2019-02-11 by nathankellenicki