nathankellenicki/node-poweredup #22: Bad color detection

Kalindor71, 2019-04-14:
I installed the Boost color sensor and found it doesn't seem to work work for me, it doesn't register all colors. Also it seems to me the light emitted from the sensor is weaker than when it's connected to the Boost set, when connected to the Boost set it does register all colors as was expected. I've replaced the batteries, but to no avail. I'm testing in quite low light conditions.

nathankellenicki, 2019-04-16:
Thanks for raising the issue here. :)

Which colors does it register and which doesn't it register? How close are you holding the sensor to the color?

Can you confirm you're on the latest hub firmware (By loading up the Powered Up app and connecting to the hub - it will update it if it's now).

And finally, to help debug, could you set the following environment variable:

"export DEBUG=lpf2hub"

Before running your code, and copy/paste what it outputs when you try to detect colors with it?

Thanks!

Kalindor71, 2019-04-21:
Hi,
I have updated both firmware and software and the problem seems to have disappeared.  To elaborate on the problem green was detected as black and blue wasn't detected all the time. If the problem reappears I will inform you again.
Thank you for you effort.

nathankellenicki, 2019-04-22:
Great - well I'm glad it resolved itself! I'll close this and hope it was just out of date firmware. :)

closed 2019-04-22 by nathankellenicki