nathankellenicki/node-poweredup #24: Disconnect event not firing when hub turned off

mark-forza, 2019-04-21:
I'm trying to display the powered up network topology graphically, but when I turn off a hub there is no event that fires to indicate that a hub has disappeared. Based on my reading of the code it looks like there should be a disconnect event, but I don't see it. Is there a way to be notified of hub disconnection?

nathankellenicki, 2019-04-21:
Hi Mark, there should indeed be a disconnect event. How are you turning the hub off?

mark-forza, 2019-04-21:
I'm turning it off by holding down the power button on the hub. The light starts blinking then it's off

nathankellenicki, 2019-04-21:
What platform/OS/node version/Bluetooth adapter are you using?

nathankellenicki, 2019-04-21:
For reference, here is an example of using the "disconnect" event: https://github.com/nathankellenicki/node-poweredup/blob/master/examples/leds.js

I've just tested here with a PUP hub and it works - I'm using a MacBook Pro with built-in Bluetooth, and macOS 10.14.3.

mark-forza, 2019-04-21:
I have an Acer notebook with built-in bluetooth running Fedora Linux 29. I'm using Node 10, which required a patched version of bluetooth-hci-socket to get it to compile. My example program is like yours except it reads the colour sensor. Turning off the hub shows no disconnection in the console log. Is there any way to get more debugging information?

mark-forza, 2019-04-21:
When I run my test program and bluetoothctl simultaneously, I can see in bluetoothctl that the device is disconnected:

```[mark@localhost src]$ bluetoothctl 
Agent registered
[NEW] Device 90:84:2B:03:4F:FC Smart Hub
[CHG] Device 90:84:2B:03:4F:FC Connected: yes
[CHG] Device 78:BD:BC:99:DD:D0 RSSI: -74
[CHG] Device 90:84:2B:03:4F:FC Connected: no
[DEL] Device 90:84:2B:03:4F:FC Smart Hub
```

But in the test program I don't see any console logs for that.

nathankellenicki, 2019-04-21:
I'm doing some testing myself and am experiencing this behaviour on another device of mine. It seems that in some circumstances either the `disconnect` event isn't emitted or caught for some reason.

Further debugging can be done by setting a `DEBUG` environment variable, as the Noble library in use here emits debug information to it. Setting `DEBUG=*` shoud log everything, but I don't see much debug info in there.

I'll continue to debug!

nathankellenicki, 2019-04-21:
I think this might be a scoping issue. Seems the Noble peripheral needs to maintain scope with noble for it to receive events, but I think it's losing it somewhere and being bound to the poweredUp instance. So the disconnect event never gets triggered.

nathankellenicki, 2019-04-25:
Turns out it wasn't that at all. Was merely that when I discover a device, I remove all existing event handlers first, as Noble can emit the same peripheral instance for multiple discoveries of the same device. Turns out I was just removing those event handlers too late, and instead should have done it before I started initialising the LPF2 hub.

Fixed in v2.4.1.

closed 2019-04-25 by nathankellenicki

mark-forza, 2019-04-26:
Works now for me, thanks!