nathankellenicki/node-poweredup #26: Does not connect to the hub?

cfelxleduc, 2019-08-05:
Hi, 

I have the cargo train, I'm trying to connect my raspberrypi with the hub to control the train. 
I'm trying the Node.js Sample Usage, but it stay at Scanning for Hubs. I'm new with Node Js, im not sure how could I solved that problem? 

Thanks

nathankellenicki, 2019-08-05:
Hi, thanks for the message. :) I'm going to need some more info to help though.

1. What platform? (macOS, Windows, Raspbian, etc)
2. Which version of Node.js are you running?
3. Can you run `export DEBUG=*` and try it again and paste the output?

Thanks,
Nathan

cfelxleduc, 2019-08-05:
Hi Nathan, 
Here the information 
1. Raspbian GNU/Linux 10 (buster)
2. Node version is v8.0.0
3. I'm not sure how to run export DEBUG=*, could you give me more information? 

[image]
[image]

Thanks

nathankellenicki, 2019-08-05:
So from your screenshot above you wrote the commands `ls` then `sudo node lego.js`. In between, can you run `export DEBUG=*`

nathankellenicki, 2019-08-05:
Also just to clarify, which model of Raspberry Pi are you running? If it's an RPi 2 or below, do you have some kind of Bluetooth adapter plugged in? Or are you running an RPi 3.

cfelxleduc, 2019-08-05:
I'm running on a Raspberry pi 3 version a02082 and yes im pressing the green button after i see the scanning message. 

[image]

nathankellenicki, 2019-08-05:
Interesting there's no DEBUG output after running your code. This suggests to me the the module isn't installed properly - I also don't see a package.json in your directory. How did you initialize your project?

Can I ask you to do this:
1. Create a new directory.
2. Run `npm init -y` in this new directory.
3. Run `npm install node-poweredup --save` in there.
4. Copy your lego.js script into here.
5. Run `export DEBUG=*`
6. Then run it with `sudo node ./lego.js`.

cfelxleduc, 2019-08-05:
I did everything you just said, it's still a the same place. 
The light on my hub is flashing, and on the terminal is still " Scanning for hub "

cfelxleduc, 2019-08-05:
[image]

nathankellenicki, 2019-08-05:
Oh my apologies, I forgot you're running the script under sudo. 🤦‍♂ Can you run `sudo DEBUG=* node leds.js`

cfelxleduc, 2019-08-05:
No problem, thanks for your help. 
Yes I confirm that I install the Node prerequisites. 

Here what I got:
pi@raspberrypi:~/ControlTrain $ sudo DEBUG=* node ./lego.js
Scanning for Hubs...
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +13ms
  hci set le event mask - writing: 010120081f00000000000000 +5ms
  hci read local version - writing: 01011000 +1ms
  hci write LE host supported - writing: 016d0c020100 +1ms
  hci read LE host supported - writing: 016c0c00 +1ms
  hci read bd addr - writing: 01091000 +1ms
  hci onSocketData: 040e0401010c00 +8ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +1ms
  hci           cmd = 3073 +1ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0401012000 +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8193 +1ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0c01011000076801070f000922 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4097 +1ms
  hci           status = 0 +0ms
  hci           result = 076801070f000922 +0ms
  hci set scan enabled - writing: 010c20020001 +1ms
  hci set scan parameters - writing: 010b200701100010000000 +1ms
  hci onSocketData: 040e04016d0c00 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3181 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e06016c0c000100 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3180 +0ms
  hci           status = 0 +0ms
  hci           result = 0100 +0ms
  hci                   le = 1 +1ms
  hci                   simul = 0 +0ms
  hci onSocketData: 040e0a01091000dbb541eb27b8 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4105 +1ms
  hci           status = 0 +0ms
  hci           result = dbb541eb27b8 +0ms
  hci address = b8:27:eb:41:b5:db +2ms
  noble addressChange b8:27:eb:41:b5:db +1ms
  hci onSocketData: 040e04010c200c +0ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010b2000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8203 +0ms
  hci           status = 0 +0ms
  hci           result =  +1ms
  noble stateChange poweredOn +1ms
  poweredup Scanning started +0ms
  hci set scan enabled - writing: 010c20020001 +6ms
  hci set scan parameters - writing: 010b200701100010000000 +0ms
  hci set scan enabled - writing: 010c20020101 +1ms
  hci onSocketData: 040e04010c200c +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010b2000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8203 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010c2000 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  noble scanStart +1ms
  hci onSocketData: 043e2b020103017f1ba7f6614d1f1eff060001092002c172678b17fc566846f18f308d733cab073a229ff0ed60c8 +8ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +1ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 03017f1ba7f6614d1f1eff060001092002c172678b17fc566846f18f308d733cab073a229ff0ed60c8 +0ms
  hci                   type = 3 +2ms
  hci                   address = 4d:61:f6:a7:1b:7f +0ms
  hci                   address type = random +0ms
  hci                   eir = 1eff060001092002c172678b17fc566846f18f308d733cab073a229ff0ed60 +0ms
  hci                   rssi = -56 +1ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[6,0,1,9,32,2,193,114,103,139,23,252,86,104,70,241,143,48,141,115,60,171,7,58,34,159,240,237,96]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +3ms

nathankellenicki, 2019-08-06:
Is that everything in the debug log? It looks like it scanned and found some kind of device but its not a known Powered UP hub. The manufacturerData for a Powered UP hub should be `[ 151, 3, 0, 65, 7, 178, 99, 0 ]`.

cfelxleduc, 2019-08-07:
I did it again, here the output. It's seem he can't see the Powered UP as the data should be [ 151, 3, 0, 65, 7, 178, 99, 0 ] but if I open the bluetooth tools, he can see the LEGO Bootloader. 

It seem that my raspberry pi cant see my HUB

pi@raspberrypi:~/ControlTrain $ sudo DEBUG=* node ./lego.js
Scanning for Hubs...
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +17ms
  hci set le event mask - writing: 010120081f00000000000000 +5ms
  hci read local version - writing: 01011000 +2ms
  hci write LE host supported - writing: 016d0c020100 +1ms
  hci read LE host supported - writing: 016c0c00 +1ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0401010c00 +9ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 3073 +1ms
  hci 		status = 0 +1ms
  hci 		result =  +0ms
  hci onSocketData: 040e0401012000 +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8193 +0ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e0c01011000076801070f000922 +0ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4097 +0ms
  hci 		status = 0 +1ms
  hci 		result = 076801070f000922 +0ms
  hci set scan enabled - writing: 010c20020001 +2ms
  hci set scan parameters - writing: 010b200701100010000000 +2ms
  hci onSocketData: 040e04016d0c00 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +1ms
  hci 		cmd = 3181 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +1ms
  hci onSocketData: 040e06016c0c000100 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 3180 +1ms
  hci 		status = 0 +0ms
  hci 		result = 0100 +0ms
  hci 			le = 1 +0ms
  hci 			simul = 0 +0ms
  hci onSocketData: 040e0a01091000dbb541eb27b8 +1ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +0ms
  hci 		result = dbb541eb27b8 +0ms
  hci address = b8:27:eb:41:b5:db +3ms
  noble addressChange b8:27:eb:41:b5:db +2ms
  hci onSocketData: 040e04010c200c +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8204 +0ms
  hci 		status = 12 +0ms
  hci 		result =  +0ms
  hci onSocketData: 040e04010b2000 +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 8203 +1ms
  hci 		status = 0 +0ms
  hci 		result =  +0ms
  noble stateChange poweredOn +3ms
  poweredup Scanning started +0ms
  hci set scan enabled - writing: 010c20020001 +12ms
[... 273 lines cut ...]
  hci onSocketData: 043e2b020102017d1a3ce12f561f1eff4c000719010f2036aa7518001068cd04b00dfd385540092903a5b09b81ab +1s
  hci 	event type = 4 +0ms
  hci 	sub event type = 62 +1ms
  hci 		LE meta event type = 2 +1ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 02017d1a3ce12f561f1eff4c000719010f2036aa7518001068cd04b00dfd385540092903a5b09b81ab +1ms
  hci 			type = 2 +4ms
  hci 			address = 56:2f:e1:3c:1a:7d +0ms
  hci 			address type = random +1ms
  hci 			eir = 1eff4c000719010f2036aa7518001068cd04b00dfd385540092903a5b09b81 +0ms
  hci 			rssi = -85 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,7,25,1,15,32,54,170,117,24,0,16,104,205,4,176,13,253,56,85,64,9,41,3,165,176,155,129]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +0ms
  hci onSocketData: 043e2b02010301bac4fda03e441f1eff4c0007190102200b880f010000a5c445caf6e83d83bda3a519fba102fcae +4s
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +0ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 0301bac4fda03e441f1eff4c0007190102200b880f010000a5c445caf6e83d83bda3a519fba102fcae +1ms
  hci 			type = 3 +0ms
  hci 			address = 44:3e:a0:fd:c4:ba +0ms
  hci 			address type = random +0ms
  hci 			eir = 1eff4c0007190102200b880f010000a5c445caf6e83d83bda3a519fba102fc +0ms
  hci 			rssi = -82 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,7,25,1,2,32,11,136,15,1,0,0,165,196,69,202,246,232,61,131,189,163,165,25,251,161,2,252]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +1ms
^C  hci set scan enabled - writing: 010c20020001 +4s

cfelxleduc, 2019-08-09:
Can I force the connexion to that manufacturerData?

dlech, 2019-08-09:
> he can see the LEGO Bootloader.

If you are seeing the name "LEGO Bootloader" then either A: you are pressing and holding the button for more then 5 seconds when powering on or B: you have bad firmware and need to use the official app to restore the firmware (or C: you actually renamed your brick this, but that doesn't seem likely :smile:).

nathankellenicki, 2019-08-10:
> Can I force the connexion to that manufacturerData?

Currently not without discovering it. Its very strange that your device isn't discovering it though.

Can you try updating the hub's firmware from the Powered Up hub? Can you successfully connect to it from that?

Do you have another device you can try from, such as a Mac or something? Otherwise I'll have to dig one of my RPi 3's out. :)

nathankellenicki, 2019-11-10:
Closing due to inactivity.

closed 2019-11-10 by nathankellenicki