nathankellenicki/node-poweredup #27: Impossible to disconnect

ebouda33, 2019-08-09:
hi
when i disconnect the hub
poweredUp remove the hub
but the legohub is still connected to my computer.
i use this.poweredUP.getConnectedHubByUUID(decodeURIComponent(uuid)).disconnect()

i m under angular and i use your library.
MacOS 10.14.6
i do not  have any worries  with speed or change led or connect to.

nathankellenicki, 2019-08-10:
Interesting! I haven't seen this before. What type of hub is it? Can you try `export DEBUG=*` and then running the code again and see what it says when disconnecting?

ebouda33, 2019-08-10:
where I put the code `export DEBUG=*` 
in angular I don't see where I write it!!!
debug function of your library

for information shutdown work correctly

nathankellenicki, 2019-08-13:
Aha, my bad - I just realised you're in the browser not Node.js. I'll have to do some testing and get back to you!

nathankellenicki, 2019-08-13:
Does shutdown do what you need right now?

nathankellenicki, 2019-11-10:
I'm unable to reproduce this using the `examples/web_bluetooth.html` example. Are you able to repro this using that example?

aileo, 2019-11-20:
Hi,
To enable debug log browser side, just run `localStorage.debug = '*'` in the javascript console.

nathankellenicki, 2019-12-03:
Closing due to inactivity (112 days).

closed 2019-12-03 by nathankellenicki