nathankellenicki/node-poweredup #28: Uncaught (in promise) DOMException

jwbaart, 2019-08-12:
When using the browser version and connecting on a Android Pixel 2 in Chrome I'm getting the following error 'Uncaught (in promise) DOMException'. When connecting on a laptop it works.

[image]

This relates to the following function:
`
writeToCharacteristic(uuid, data, callback) {
        this._queue = this._queue.then(() => this._characteristics[uuid].writeValue(data)).then(() => {
            if (callback) {
                callback();
            }
        });
    }
`

nathankellenicki, 2019-08-13:
Interesting...this might be a race condition I'll need to look into. I have a sneaking suspicion as to the cause but won't be able to test it. Let me get back to you!

jwbaart, 2019-08-14:
:thumbsup:
Let me know if you need any more information or a url to test.

nathankellenicki, 2019-09-08:
Apologies, I've been on vacation the past few weeks. Do you have some sample code that causes this to trigger?

jwbaart, 2019-09-09:
No problem! :)

The issue occurs [here](https://duplo-train.jwbaart.dev/) (when connecting to a train). Code can be found [here](https://github.com/jwbaart/duplo-train/blob/master/store/duplo.js#L45)

nathankellenicki, 2020-02-10:
I was unable to reproduce this, likely due to not having the correct devices. Is this still an issue? (Web Bluetooth is still in its infancy so it wouldn't surprise me if each device and browser behaved differently/weirdly).

nathankellenicki, 2020-05-23:
Closing due to inactivity.

closed 2020-05-23 by nathankellenicki

Gwened, 2020-10-17:
Also got this exception on a Android phone (while it was fine on a Mac laptop): "**DOMException: GATT operation failed for unknown reason**".
I got inspired by [that issue](https://github.com/thegecko/web-bluetooth-dfu/issues/28) and fixed it successfully by delaying a bit the call to writeCharacteristic in _determineLPF2HubType():
```
	setTimeout(() => {
		device.writeToCharacteristic(Consts.BLECharacteristic.LPF2_ALL, Buffer.from([0x05, 0x00, 0x01, 0x0b, 0x05]));
}, 50);
```

kunaakos, 2021-02-05:
Getting the same issue on an android phone (Google Pixel 3a), same build works fine on desktop Chrome. Could this issue be reopened @nathankellenicki? I'm working on a lego remote web app that's pretty pointless if I can't get it working on touch devices :(