nathankellenicki/node-poweredup #29: UnhandledPromiseRejectionWarning with Vernie Roaming Example

mharmuth, 2019-08-14:
I always get the same UnhandledPromiseRejectionWarning when using the vernie_roaming.js example. Here is the full output:

```
$ >  node app.js
Looking for Vernie...
Connected to Vernie! Battery level: 100
(node:50807) UnhandledPromiseRejectionWarning: Error: Angle rotation is only available when using a Boost Tacho Motor, Boost Move Hub Motor, Control+ Medium Motor, or Control+ Large Motor
    at BoostMoveHub.setMotorAngle (/path/lego-boost/node_modules/node-poweredup/dist/node/boostmovehub.js:172:19)
    at turnHeadRight (/path/lego-boost/app.js:29:18)
    at /path/lego-boost/app.js:52:15
    at new Promise (<anonymous>)
    at scan (/path/lego-boost/app.js:47:12)
    at Timeout._onTimeout (/path/lego-boost/app.js:129:80)
(node:50807) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:50807) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

What's strange about it: Vernie starts to move backwards in the beginning and then without hitting anything the exception is thrown. I am using the standard LEGO Boost Set with nothing else.

I am using macOS 10.14.6 and node 12.6.0.

nathankellenicki, 2019-08-16:
This typically happens when the hub is slow to report what motors/sensors are attached to what ports. Sometimes there can be a little bit of a delay, and I've noticed this has increased with the latest firmware.

Try increasing the 1000ms sleep to something larger, like 3000ms, and let me know what happens.

nathankellenicki, 2019-12-03:
Closing due to inactivity (109 days).

closed 2019-12-03 by nathankellenicki