nathankellenicki/node-poweredup #30: Can't connect to Control+ hub

tilmx, 2019-08-25:
Hey Nathan!

First of all – super nice project. Thank you so much!
I tried to connect to connect a Control+ hub on my MacBook with Node, but wasn't able to to do that. I'm running on MacOS 10.14.6 and Node 8.0.0/10.16.3/12.9.0 (none of them worked). Running `DEBUG=* node ./example.js` just says:
```
Scanning for Hubs...
  noble stateChange poweredOn +0ms
  poweredup Scanning started +0ms
  noble scanStart +1ms
```
and then nothing happens.

Interestingly, the connection works fine via the Web API.

Do you have an idea what I could try or where I could dig deeper?

ronbenetti, 2019-08-29:
The web api is now working with the technic hub and my Mac. Not sure if it was a re-boot or something else. 

I have not tested the Node version.

tilmx, 2019-09-03:
My little node project suddenly worked now. Not sure why it couldn't connect for the last weeks, but now it does perfectly. Thanks nevertheless! :-)

closed 2019-09-03 by tilmx

nathankellenicki, 2019-09-08:
Apologies, I've been on vacation in South America the past few weeks, but I'm glad the issue was (somehow) resolved. :)