nathankellenicki/node-poweredup #32: Is it possible to read tacho motor from a Control+ Smart hub?

jabrena, 2019-10-28:
Hi,

I would like to know if exist a way to read the tacho motor value from a Motor connected to a Control + Smart hub.

[image]

Many thanks in advance

Juan Antonio

nutki, 2019-11-04:
I don't have one to test, but you should be able to do `cplushub.on("rotate", (port, position) => console.log(port, position))`.

nathankellenicki, 2019-11-10:
Yep, you can check the rotation value by listening to "rotate" events. I'm still looking to implement a way of reading the angle from-zero.

derammo, 2019-12-02:
that would be interesting if you can read angle from zero, since the motor acts like it doesn't have an absolute zero point (see #58)  

if you got this part to work, we could also implement setAbsolutePosition based on reading the current position, which would be sweet

aileo, 2019-12-03:
Third mode of L/XL motors is `APOS` which I understand as "absolute position". You should be able to get this absolute position by listening to rotate event and subscribing to the motor with mode `0x03`.

nathankellenicki, 2019-12-03:
That's correct. Absolute positioning is currently not fully supported, but hopefully soon. In the meantime you can use the existing mode.

closed 2019-12-03 by nathankellenicki