nathankellenicki/node-poweredup #38: Question : Get Hub's LED color

aileo, 2019-11-03:
Hello,

First, thank you for this awesome project.
I was wondering if there is a way to get the current colors of the hub built-in LED ?

Thank you.
Léo

nutki, 2019-11-05:
This is not implemented as far as I know, though maybe technically possible. But either way the led color resets on connection and there seem to be no way to change it outside of this library, so why not simply storing the last used value used on the client side?

aileo, 2019-11-07:
Thank you for the answer.
I use it in a webpage with no persistency between connections, I was trying to identify visually the hubs by their LED colors.
I don't like to store the LED color in the app state because it would be like storing twice the same value on two asynchronous systems without any guarantee about the real state (on device).
However, I actually do it this way for now but I was hoping for a "better" solution.

Note: it is the first project of this kind I try to play with, so a lot of my initial thoughts are probably wrong...

nutki, 2019-11-07:
I see. But doesn't the LED color reset to white at each reconnect? (So you will not be able to keep the color between the two pages anyway).

aileo, 2019-11-07:
Yes, it resets color between connections (white for PoweredUp and blue for Boost and Control+).
Should I close this issue or lets it opened as a feature request?

nathankellenicki, 2019-11-10:
I think this is possible, but if you want to use the LED color to identify the hubs, I'd recommend either going by the hub name or type instead.

You can set each hub's unique name with `setName("My Name Here");` and read it from the `name` property.

aileo, 2019-11-14:
I think that reading values from devices (led/motor/sensor) should be discussed in #49 so let's close this one.

closed 2019-11-14 by aileo