nathankellenicki/node-poweredup #39: Should the control+ hub have a TILT port like the boost move hub ?

aileo, 2019-11-04:
Hi,

Should the control+ hub have a TILT port like the boost move hub ?

I had a little look at the code to understand and could not figure out why this line is commented :
https://github.com/nathankellenicki/node-poweredup/blob/master/src/controlplushub.ts#L38

Thank you,
Léo.

nutki, 2019-11-05:
The port object is just for internal use and optional for build-in devices. From the API point of view the 'ports' are just strings, and the tilt port of the control plus hub is used here https://github.com/nathankellenicki/node-poweredup/blob/master/src/lpf2hub.ts#L426

nutki, 2019-11-05:
Also note that the commented line is using 60 (0x3c) for the port number while the code that reads the accelerometer is using 99 (0x3c). I don't have c+ hub, so I am not sure which is correct.

aileo, 2019-11-07:
Thank you for the response. When I connect my hubs, I don't which one is discovered, so I rely on the `_port` to get all featured motors and sensors to determine which is which.

I assumed it was a bad idea regarding the "_" prefix, but I can't think of another way to do it.
Any thought ?

nutki, 2019-11-07:
You could rely on hub name or type to detect the existence of internal devices. Connectable ports will always be present in the `_port` array since this is where the library stores the info about the current device connected.

aileo, 2019-11-07:
Sounds fair to me.
I still have a question then, just by curiosity : Why does the Boost move hub have this `TILT` port in `_ports` but not the Control+ hub as it is built-in sensor for both?

note : name is not reliable as I have two PUP hubs which have different names and I did not change it myself.

nutki, 2019-11-07:
I am not the author, but if you look at `_parseSensorMessage` function, you will note that some builtin sensors are handled at the beginning before the call to `_getPortForPortNumber`: https://github.com/nathankellenicki/node-poweredup/blob/master/src/lpf2hub.ts#L372
Anything handled there (which includes voltage, current, rssi and control+ sensors) does not require the port object. Even though all those sensors (unlike for example the green button and battery level indication) have ports assigned at the LPF2 protocol level.

nathankellenicki, 2019-11-10:
It's a good question, and the question alone raises a couple of issues with how I've organised the sensor parsing code in the library.

I actually think the question should be reversed. Should the BOOST Move Hub have a TILT port on it? In my library I've tried to abstract away internal ports, exposing them as features. In an ideal world users would have no knowledge of the concept of internal/non-pluggable ports, with each hub just having a feature set.

This however ignores the issues of the external tilt sensor that can be plugged into the hub, and so we need to tell the user which port the tilt notification came from, which unfortunately makes the answer to the previous question "yes". So for consistency, I'd agree, the Control+ hub should have a TILT port. I'll work on this when I have access to my Control+ hub again. :)

However, @aileo, if you're looking for a way to uniquely identify hubs, I'd recommend using something unique rather than going off the port layout. You could use the `uuid` of the hub, or you could give each one a name using `setName()`. Or if you need to identify the type of hub, check the `type` property of check the type of the Hub object itself.

aileo, 2019-11-11:
Thinking about the port/feature, I can see three types of component:
- actuators (motors, leds)
- sensors to play with, which you somehow have control on (tilt, color, distance)
- system sensors (battery, current, voltage, rssi)

There is already two "port" objects in hubs (`_ports` and `_virtualPorts`). I have the feeling that adding a third `internalPorts` would simplify as everything would be managed as port/device internally.
Here are the ports I would get in the `internalPorts`:
- BoostMoveHub: GREEN, LED, A, B, TILT
- PUPHub: GREEN, LED
- PUPRemote: GREEN, LED, LEFT, RIGHT
- ControlPlusHub: GREEN, LED, TILT, CPU
I don't have DuploTrainBase or WeDo2SmartHub so I can't tell...

I noticed the Boost App allow to set the color&distance sensor led color, I wonder if this approach can help to handle this use case as `setLED*` method could then accept a port parameter too.

I don't understand the whole thing yet so that was just my two cents.

Using port layout as an identification method was only to mimic the control plus app behavior which I assume does this to identify both hubs on the 42100.

dlech, 2019-11-11:
FYI, although the IMU on the Control+ hub is a single device physically, in the communication protocol, it is treated as 5 different sensors on 5 different ports. It is quite different from the BOOST hub.

nathankellenicki, 2019-11-11:
So I had more thoughts about it last night and I think the combination of `_ports` and `_virtualPorts` isn't quite right. I think from a user perspective, they shouldn't care which devices (sensors or motors) are internal or external - they are just devices. I think on the other hand, things such as the voltage or current sensor are different - not something a user would think of as a device like a classical motor or sensor.

I've had a plan in my head for a little while now (which I've had little time to execute on!) to remove all motor or sensor control from the hub class itself, and move them to individual classes, inherited from a base device class. This would prevent the code for the base hub device from getting too large (it already kinda is) as more sensors and motor types are added, as well as make it easier to allow sensor mode switching (if the functionality for each sensor is contained within its own class).

In the meantime though, I think the original request in this thread is valid. I've made the Control+ TILT (and the ACCEL accelerometer) behave like normally attached devices. So they show up in as normal ports and have modes set like normal ports. It also removed the special message parsing for these ports and became part of the normal flow - https://github.com/nathankellenicki/node-poweredup/compare/v4.1.0...v4.2.0

nathankellenicki, 2019-11-11:
> Thank you for the response. When I connect my hubs, I don't which one is discovered, so I rely on the `_port` to get all featured motors and sensors to determine which is which.
> 
> I assumed it was a bad idea regarding the "_" prefix, but I can't think of another way to do it.
> Any thought ?

Regarding this, would you like a public (_ is private) way of quering which ports are available?

nathankellenicki, 2019-11-11:
> Using port layout as an identification method was only to mimic the control plus app behavior which I assume does this to identify both hubs on the 42100.

This is true, it does do this. I assume it does this though as it can't rely on names or uuid's for the hubs, since the app doesn't know which names or uuids to use for each hub.

The benefit of us being able to write code is that we can define which values to look for, so we don't need to implement such a hack. :)

aileo, 2019-11-12:
> In the meantime though, I think the original request in this thread is valid. I've made the Control+ TILT (and the ACCEL accelerometer) behave like normally attached devices. So they show up in as normal ports and have modes set like normal ports. It also removed the special message parsing for these ports and became part of the normal flow - [v4.1.0...v4.2.0](https://github.com/nathankellenicki/node-poweredup/compare/v4.1.0...v4.2.0)

A big thank you !

> Regarding this, would you like a public (_ is private) way of quering which ports are available?

Yes, in fact, available ports are some kind of static configuration. I think ports of each hub type could be in Consts, then `hub.getPortDeviceType(port)` can do the trick. It would also simplify map/reduce over ports (for device listing or else).

I think of something like:
```js
{
  [Consts.HubType.POWERED_UP_HUB]: ['A', 'B'],
  [Consts.HubType.POWERED_UP_REMOTE]: ['LEFT', 'RIGHT'],
}
```

> I've had a plan in my head for a little while now (which I've had little time to execute on!) to remove all motor or sensor control from the hub class itself, and move them to individual classes, [...]

Would you like to discuss this topic in a dedicated issue ? I am not very confident with typescript and the protocol but I would be happy to help.

> The benefit of us being able to write code is that we can define which values to look for, so we don't need to implement such a hack. :)

I  work on two use cases that don't match this statement:
- A flexible interface to attach visual controls to devices, so my 7yo can play without coding (at least for now)
- An simple interface "generator" for MOC creators with some kind of device mapping (like for the 42100)

nutki, 2019-11-12:
> So I had more thoughts about it last night and I think the combination of `_ports` and `_virtualPorts` isn't quite right. I think from a user perspective, they shouldn't care which devices (sensors [...]

I agree that user API should be as simple as possible and maintaining device concept directly reflecting the BT interface is not the best approach. In some cases some abstraction of this kind is useful as motors, sensors, and hub external ports are physical objects and the library users will expect they will be represented with some library object as well.

On the other hand internally having port and device objects exactly matching the concept of port and device of the bluetooth interface is a good thing, and I would encourage to keep it. For example the current port mode is not reported back with the port value messages so it has to be remembered in the client to properly parse those. A port/device object is a good place for it. Similarly with port value ranges.

Here is my idea of an API which would help both the library API users and code decomposition of the library itself would be like this:

Now:
```js
const horizonExpress = {
  hub: horizonPupHub,
  motorPort: "A",
  lightsPort: "B"
}
...
horizonExpress.hub.setMotorSpeed(horizonExpress.motorPort, -50);
horizonExpress.hub.setLightBrightness(horizonExpress.lightsPort, 100);
...
horizonExpress.hub.brakeMotor(horizonExpress.motorPort);
horizonExpress.hub.setLightBrightness(horizonExpress.lightsPort, 0);
```
After:
```js
const horizonExpress = {
  motor: horizonPupHub.getMotor("A"),
  lights: horizonPupHub.getLight("B"),
}
...
horizonExpress.motor.speed(-50);
horizonExpress.lights.brightness(100);
...
horizonExpress.motor.brake();
horizonExpress.lights.off();
```

aileo, 2019-11-12:
I like @nutki proposal, I would had some abstraction on device type to be more permissive, a bit more like the `getConnectedHubByUUID`

```js
const heA= horizonPupHub.getDeviceByPort("A");
const heB= horizonPupHub.getDeviceByPort("B");

const horizonExpress = {
  motor: heA.type === DeviceType.TRAIN_MOTOR ? heA : heB,
  lights: heA.type === DeviceType.LED_LIGHTS ? heA : heB,
}
...
horizonExpress.motor.speed(-50);
horizonExpress.lights.brightness(100);
...
horizonExpress.motor.brake();
horizonExpress.lights.off();
```

We could also imagine some kind of `getDevicesByType` returning an array of devices.

nathankellenicki, 2019-11-12:
This is very similar to what I was thinking - I was actually just thinking of `getDevice("A")` (though maybe `getDeviceByPort` is more descriptive), which would retrieve whatever object is there. And you can determine its type by either doing `typeof device` (Since all devices would be derived from a Device base class), or `device.type`.

nathankellenicki, 2019-11-14:
Since the discussion has progressed past the original topic (which has been solved - the TILT port has been added), I'm going to close this and open a new issue based around what we're currently discussing. :)

closed 2019-11-14 by nathankellenicki