nathankellenicki/node-poweredup #43: Cannot find module

ORANGEID, 2019-11-09:
Hi

I having problems running the code, It errors with

```
internal/modules/cjs/loader.js:302
      throw err;
      ^

Error: Cannot find module '/Projects/train_controller/duplo-app_controller/dist/node/index-node.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (internal/modules/cjs/loader.js:294:19)
    at Function.Module._findPath (internal/modules/cjs/loader.js:525:18)
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:781:27)
    at Function.Module._load (internal/modules/cjs/loader.js:687:27)
    at Module.require (internal/modules/cjs/loader.js:849:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/Projects/train_controller/duplo-app_controller/test3/node.js:1:19)
    at Module._compile (internal/modules/cjs/loader.js:956:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:973:10)
    at Module.load (internal/modules/cjs/loader.js:812:32) {
  code: 'MODULE_NOT_FOUND',
  path: '/Projects/train_controller/duplo-app_controller/package.json',
  requestPath: '..'
}

```

I installed by doing the following

npm init -y
npm install node-poweredup --save
sudo node ./node.js

I am running OSX 10.14.5
It managed to make it work on my work machine running OSX 10.12

So it there something i need to change to make it run on 10.14

Any help would very much be appreciated

Cheers

James

ORANGEID, 2019-11-10:
I managed to fix this by removing the node and npm installs and reinstalling

I am connecting now
and can control the sound and colour however when ever i try to control the motor with hub.setMotorSpeed("A", 100,  2000) is errors with

Port A does not exist on this Hub type, I also tried using "B" and "C"

I am using a mac 10.14 with a 'LEGO® DUPLO® 10874 Steam Train

Any help would be great

Cheers

James

nathankellenicki, 2019-11-10:
Hi Jamie, on the Duplo train hub, the motor port is named "MOTOR". Try:

`hub.setMotorSpeed("MOTOR", 100, 2000);`

ORANGEID, 2019-11-10:
Thankyou that did the trick :)

closed 2019-11-10 by nathankellenicki