nathankellenicki/node-poweredup #46: hub.uuid changes

aileo, 2019-11-13:
Disclaimer: I may not be directly related to this module.

Using the same build of my app over two urls (localhost / 127.0.0.1), the same PUP Hub get different uuid.

From what I observed, doing multiple re-connection between both url, the uuid persists per url.

Any idea of a persitent way to identify hub ?

aileo, 2019-11-13:
**SPOILER** : it is not related to the module.

Made some research and it seems to be a feature from the web bluetooth API to protect privacy : https://webbluetoothcg.github.io/web-bluetooth/#remote-device-identifiers

From what I understand, unless LEGO add some sort of serial number to hubs, it won't be possible to identify accurately a hub between connections...

dlech, 2019-11-13:
Once connected to the hub, you can read the (unique) Bluetooth address via the LWP3 protocol.

See https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#hub-property-and-operation

"Primary MAC address" is the Bluetooth address of the hub normally and "Secondary MAC address" is the Bluetooth address of the hub when it is in firmware update mode.

aileo, 2019-11-13:
Thank you @dlech, it looks like I was a bit too pessimistic closing this.

It does not seems to be implemented yet here or I did not find it.

nathankellenicki, 2019-11-13:
Good find! As mentioned, the UUID changing is a Web Bluetooth privacy thing (iOS and macOS do it too natively), but I was unaware the hub was capable of reporting its own MAC address.

One would argue its probably a similar privacy violation for it to do so, but that's Lego's problem, not ours. :) Thanks for the PR!

closed 2019-11-13 by nathankellenicki