nathankellenicki/node-poweredup #53: Attach event emitted only if autoSubscribe

aileo, 2019-11-20:
Code from [hub.ts, line 280 to 289](https://github.com/nathankellenicki/node-poweredup/blob/e96b7bd47df345dcba9a26648b516e7467e63ecc/src/hub.ts#L280), only emit the `attach` event only if `autoSubscribe` is `true` :
```ts
if (this.autoSubscribe) {
    this._activatePortDevice(port.value, type, this._getModeForDeviceType(type), 0x00);
    /**
     * Emits when a motor or sensor is attached to the Hub.
     * @event Hub#attach
     * @param {string} port
     * @param {DeviceType} type
     */
    this.emit("attach", port.id, type);
}
```

While `detach` event is emitted whatever the value of `autoSubscribe`.
As there is no possible subscribe to get back `attach` events, I wonder if it should emit them in any case ?

nathankellenicki, 2019-11-20:
You should be able to get back attach events by calling `subscribe()` on the hub for the ports/devices you want.

nathankellenicki, 2019-12-03:
Closing as PR approved and merged.

closed 2019-12-03 by nathankellenicki