nathankellenicki/node-poweredup #58: BOOST_TACHO_MOTOR and setAbsolutePosition

derammo, 2019-12-01:
I have an 88008 V46 connected to a Boost Move Hub, and I am able to set it to absolute angles with the Lego Powered Up app.   This library disallows this as unsupported, since the motor reports as BOOST_TACHO_MOTOR.  To test, I hacked up boostmovehub.js to see if it would work if allowed.  For me at least, it worked fine.

Maybe this is supported but the library just doesn't know it.

derammo, 2019-12-01:
ok I did find a significant limitation that might be why this is disallowed:  the motor doesn't know what angle it started in.  So whatever angle it has when it powers up is considered "0".   But it still does what you'd want from a servo otherwise.

derammo, 2019-12-01:
also: resetAbsolutePosition doesn't work.  The command appears to do nothing if sent to this motor.  It might be cheaper for me to buy a different motor at this point than spending any more time on this? :)

aileo, 2019-12-03:
The BOOST_TACHO_MOTOR does not expose an `APOS` and `CALIB` modes which I assume handle the absolution position and the absolute 0 conservation over restarts.

Using absolute positioning from the boost app resets the absolute 0 on power up to the current motor angle.

The library could allow the setAbsolutePosition for BOOST_TACHO_MOTOR but you still have to manage the offset between restarts in your application (with some calibration sequence at low power for some time).

nathankellenicki, 2019-12-03:
This is because the Boost Motor and the Control+ motors having the same hardware, but different firmwares. Although they both have the hardware for measuring angle, the Control+ motor firmware allows you to set and internally store an absolute zero position, so it can keep track and allow you to rotate to an individual location.

While this library does include a few other abstractions to get across device differences (ie. Basic motors and LED lights don't include timing hardware so start/stop commands are sent based on a client side timer), I've since started wondering if this is a good idea, and instead prefer to handle this in the application itself if desired. So I wonder the same about adding an absolute position abstraction to the Boost motor.

nathankellenicki, 2020-01-17:
Closing due to hardware differences.

closed 2020-01-17 by nathankellenicki