nathankellenicki/node-poweredup #59: Scan interrupt when a tentative to connect to a device

phantase, 2019-12-03:
Hi,
I am making some test with this library, my goal is to connect on a Raspberry Pi multiple train and multiple remote.

When I try and adapt the examples, I have the following problem: the scan for new hubs/remotes stop once a device is discovered and the connection is initiated.

I have this following test : 
```javascript
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
  
  console.log(`##############################`)
  console.log(`Discovered ${hub.name}!`);
  
  if( hub instanceof PoweredUP.PUPRemote ){
    console.log("~~ Powered UP Remote ~~")
    await hub.connect(); // Connect to the Hub
    console.log("Connected");
    await hub.sleep(3000);
    
    if(hub.name.indexOf("Remote Grey") > -1){
      await hub.setLEDColor(PoweredUP.Consts.Color.WHITE);
    } else if(hub.name.indexOf("Remote Blue") > -1){
      await hub.setLEDColor(PoweredUP.Consts.Color.BLUE);
    } else if(hub.name.indexOf("Remote Red") > -1){
      await hub.setLEDColor(PoweredUP.Consts.Color.RED);
    } else if(hub.name.indexOf("Remote Green") > -1){
      await hub.setLEDColor(PoweredUP.Consts.Color.GREEN);
    }
  }

});
poweredUP.scan(); // Start scanning for Hubs
console.log("Scanning for Hubs...");
```

Which will just color the LED of the remote depending of the name of the remote.

When I just turn on one remote, this one is connected and the LED changes color. If I turn on another remote, this remote is never discoverer.

If I turn all my remote at the same time when no other device is connected, all the remote are discovered and the connection to all the remote is made (and the LEDs' color will change).

Of course, same with standard trains hub. When one hub is connected, scan is interrupted and no more hub nor remote can be discovered.

When I disconnect the attached device, the scan restart and new devices can be discovered.

I had a look to the log (`DEBUG=* node scanners.js`) and it seems that the Noble library decide to stop scanning after the connection is enabled. After a look to the code, I don't see why it decide to stop scanning without the stop command.

Here is the (long) log: 

```
root@rpi-indoor:~/nfslocal/dev/puptest# DEBUG=* node scanners.js
Scanning for Hubs...
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +4ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +2ms
  hci write LE host supported - writing: 016d0c020100 +1ms
  hci read LE host supported - writing: 016c0c00 +1ms
  hci read bd addr - writing: 01091000 +1ms
  hci onSocketData: 040e0401010c00 +6ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3073 +1ms
  hci           status = 0 +1ms
  hci           result =  +0ms
  hci onSocketData: 040e0401012000 +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
[... 247 lines cut ...]
  poweredup Hub a4da3267aaa4 connected +367ms
  pupremote Connect completed +362ms
Connected
  hci onSocketData: 0240201600120004001b0b000f0004010137000000001000000010 +12ms
  hci   event type = 2 +0ms
  hci           cid = 4 +1ms
  hci           handle = 64 +0ms
  hci           data = 1b0b000f0004010137000000001000000010 +0ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 01 01 37 00 00 00 00 10 00 00 00 10> +5ms
  lpf2hubmodeinfo Port 01, type 0037 (POWERED_UP_REMOTE_BUTTON) +17ms
  lpf2hubmodeinfo Port 01, hardware version 1.0.00.0000, software version 1.0.00.0000 +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 21 01 01> +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 21 01 02> +1ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 0a 00 41 01 00 01 00 00 00 01> +1ms
```

I am doing my test on a Raspberry Pi 3 and a Raspberry Pi 0 with the integrated Bluetooth.
Latest Raspbian (Raspbian GNU/Linux 10 (buster))
Node: v12.13.1 for the RPi3, v11.15.0 for the RPi0
"node-poweredup": "^4.3.0"
I can give the full node dependencies if required.

I hope you will be able to give me a hint.

Thanks

nathankellenicki, 2019-12-04:
I've experienced this myself with Raspbian on the Raspberry Pi, but not on macOS. There's an open issue for it against the Noble project: https://github.com/noble/noble/issues/223. The workaround is here: https://github.com/noble/noble/issues/223#issuecomment-241924168

With this library you can call `scan()` again after connecting to the peripheral. I added this to the library at once point but it had adverse side effects, so I removed it. Hopefully it works for your scenario.

phantase, 2019-12-04:
OK, just tested and restarted the scan after the connect.

I will try to find a RPi4 to see if the problem is the same with this model.

In any case it solve my case, thanks, I will continue working/playing/testing on that.

closed 2019-12-04 by phantase

nutki, 2019-12-04:
@phantase I had the same issue on RPi3 and fixed this in my fork: https://github.com/nutki/node-poweredup/commit/6cb5c0d621e488adc3e3dbda48abe5454c2f1424
I did not have any further problems with this.

nathankellenicki, 2019-12-04:
@nutki That exact fix was implemented in this library, but I removed it as it caused a few issues in certain edge cases.

Namely:

1. Start scanning
2. Turn on two hubs at the same time
3. Library discovers one
4. Library discovers second
5. Library starts connecting to one
6. Library starts connecting to second
7. Library succeeds connecting to one, starts scanning again
8. Library discovers second again, as it is still advertising as it has not finished connecting
9. Library starts connecting to second for a second time
10. Library succeeds connecting to second
11. Repeat

I wanted to implement some kind of filtering so it wouldn't try connecting to the same device again if a connection is already in progress, which worked, but the trouble is that Noble gives each device a different peripheral id on every discover. So the original reference was no longer valid.

phantase, 2019-12-05:
Thanks for this feedback @nutki and this explanation @nathankellenicki I will probably only reactivate scan after a connect.