nathankellenicki/node-poweredup #62: Running application without sudo

superman-lopez, 2019-12-07:
I have just started using this library (thanks so much for creating and sharing!) and I run into an issue: I am only able to connect to the Boost Move Hub when I run `node` with `sudo`.  I tried to search for using Noble without `sudo` but couldn't find any results, which made me wonder if the elevated rights are a requirement from Noble or node-poweredup.  Is there anyway I can use the library without elevated rights?

The reason I would like to use run as normal user account is to facilitate debugging in VS Code.  I am running Elementary OS (Ubuntu 18.04).

nathankellenicki, 2019-12-09:
Hi! This is Noble, which requires some elevated privileges to access Bluetooth hardware. There's a few issues on Noble talking about it, perhaps one of them can solve your problem:

https://github.com/noble/noble/issues/888
https://github.com/noble/noble/issues/682
https://github.com/makebrainwaves/BrainWaves/issues/9

superman-lopez, 2019-12-09:
Ok thanks, I will look at these issues!

closed 2019-12-09 by superman-lopez