nathankellenicki/node-poweredup #66: Help

ORANGEID, 2019-12-19:
Hi there 

this isn't strictly to do with node-poweredup but i was wondering if you could give me some pointers, I trying to make a bluetooth remote using espruino and seem to fine connect to a duple train but when i try to send the buffer command it doesn't appear to be doing anything

I using the Duplo train hub this command i am trying to send
[0x08, 0x00, 0x81, 0x01, 0x11, 0x51, 0x01, 0x05]

 in your powered script i am try replicate hub.playSound(5)

this code is current using
```
function connect_train() {
  if (connected == 0) {
    NRF.connect("e0:7d:ea:0c:03:29").then(function(g) {
      console.log("Starting connecting2");
      gatt = g;
      return gatt.getPrimaryService("00001623-1212-efde-1623-785feabcd123");
    }).then(function(service) {
      return service.getCharacteristic("00001624-1212-efde-1623-785feabcd123");
    }).then(function(characteristic) {
      Characteristic_store = characteristic;
        return characteristic.readValue();
      }).then(value => {
        console.log(value);
    }).then(function() {
      console.log("Train Connected");
      connected = 1;
  });
  } else {
    gatt.disconnect();
    console.log("Train Disconnected");
    connected = 0;
  }
}

function send_message() {
  console.log("message = ", Characteristic_store);
  const sendvalue = new Uint8Array([0x08, 0x00, 0x81, 0x01, 0x11, 0x51, 0x01, 0x05]);
  Characteristic_store.writeValue(sendvalue)
  .then(_ => {
    console.log(sendvalue);
  });
}
```

any help would be greatly appreciated

Cheers

James

nathankellenicki, 2019-12-19:
If I recall correctly (I don't have my Duplo Train Base to hand), the speaker needs to be put into the preset audio output mode first. If so, you need to send this first:

`[0x0a, 0x00, 0x41, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x01]`

After that you can send your messages to play sounds.

ORANGEID, 2019-12-19:
Thankyou, thank you, thank you that has been driving me mad

I can't tell how grateful I am.

Are you signed up to one those service where i can buy you coffee as a thanks

nathankellenicki, 2019-12-23:
No problem, glad it works. I'm not on one of those services but your nice message is thanks enough. :)

closed 2019-12-23 by nathankellenicki