nathankellenicki/node-poweredup #67: Use in TypeScript application

superman-lopez, 2019-12-19:
Is it possible for me to easily use the `node-poweredup` library in a TS application?

Some superfluous information on why I ask:
After doing some playing around in and with the sample JS applications, I had intended to use the `node-poweredup` package in a TS application as I am not that familiar with vanilla JS.
After getting started the first thing I noticed was that the `.on()` was not recognized on the `PoweredUp` class.  I then noticed that `PoweredUp` is extending `EventEmitter` which is the class i suspected contains the `.on()` method.  From there I realized that NPM had not installed `events` package.  I installed it manually, however when building the application it still fails on this `.on()` due to not being able to find a value for `Promise`.

Seeing the library works fine in a JS application, I suspect that dev packages are missing but don't actually provide functionality, purely the interface definitions.  And I fear I will continue to run into these missing interfaces if I use `node-poweredup` in a TS application, making the attempt not worth the effort.

aileo, 2019-12-19:
I assume you import the built lib like this :
```js
import PoweredUP from 'node-poweredup';
```
Which does not include typescrypt definitions.

Can you try to import this way :
```js
import { PoweredUP } from 'node-poweredup/src/poweredup-node';
// or if this is a browser app
import { PoweredUP } from 'node-poweredup/src/poweredup-browser';
```

nathankellenicki, 2019-12-19:
Interesting. Can I clarify if you're trying to use this inside Node.js or a browser? I ask as the `events` package shouldn't be needed for Node.js, as `EventEmitter` is part of the standard library, so I'm going to assume this is a browser based application.

I haven't actually tried writing a browser application using this library, only Node.js, but I see what you're getting at - Webpack wouldn't bundle the typedefs for `events` in a bundled context as it transpiles to plain JS, and this library doesn't specify `events` as a dependency as it's not needed in Node.js environments (and bundled automatically by Webpack).

superman-lopez, 2019-12-20:
Thanks for your replies!  I have tried the following:
```js
import { PoweredUP } from 'node-poweredup/src/poweredup-node';
```
But that provides me the same issue on `on()`.  In fact, when I just used `node-poweredup` as source for my import, vscode (my IDE) does find the definition for `PoweredUp` in the `node-poweredup/src/poweredup-node.d.ts` file as does `ts-node` seemingly as it is not complaining about the other methods on `PoweredUp`, only about `on()`.

I am trying to build a Node.js application, not a browser based.  I can see that in the `poweredup-node.d.ts` file there is an explicit import from `events` which vscode can't resolve (and seemingly ts-node neither).  I'll research a but about this and see if I can make ts-node or Node.js resolve this import correctly.

aileo, 2019-12-20:
I thought about an other possible causes but I need to see how you instanciate PoweredUp, can you post this part of your code here ?

nathankellenicki, 2020-06-12:
This is quite old, however I only recently got around to using node-poweredup in my own Typescript application.

Aside from inadvertently exposing a WebBluetooth type (which required me having to install the type separately - I'll fix this), I didn't run into any other issues.

I uploaded the project here as an example: https://github.com/nathankellenicki/legolayout

nathankellenicki, 2020-12-23:
Closed as resolved

closed 2020-12-23 by nathankellenicki