nathankellenicki/node-poweredup #68: Getting a port value on Control+ returns 1 byte of data

mjorrens, 2019-12-21:
I have used your project as a guide to build out a full controller in .NET C#. The level of detail of your solution has been fantastic in unlocking the mysteries of the Control+ Hub. I have the Lego Liebherr 9800 which I am now controlling from my PC.

My only gotcha thus far is trying to get the rotations of a specific motor. I can reset the absolute position and go to an absolute position. All good. 

The problem is this: when I do a Port Value request for port "A": 0x05, 0x00, 0x21, 0x00, 0x00, I then get a result of 0x05, 0x00, 0x45, 0x00, 0x3a. I am expecting 4 bytes of data but I am just getting 1 byte (0x3a in this case) which seems to echo the power I am applying to the port.

In your code, a nice "rotate" event fires when a motor is turned and returns 4 bytes of data. I think I am missing something...any little bit of guidance would be great! I don't see any place in your code where you are actually registering for this "rotate" event.

My end-game here is to do a calibration on the arm of the excavator just like Lego does in their app.

mjorrens, 2019-12-21:
PROBLEM SOLVED. In reviewing your code, you activate all Control+ ports with mode 0x02 via command 0x41. I was not doing that. Now, I do get the 4 bytes in return message 0x45 of the rotation.

Doing a straight Port Value request with command 0x21 and mode 0x00 returns the current power level applied to the motor. Doing a Port Value request with mode 0x02, returns other static information. Anyway, that is where I got stuck thinking the outputs for 0x21 and 0x41 with the same mode info would do the same thing.

closed 2019-12-21 by mjorrens

tthiery, 2020-06-28:
@mjorrens Is you source code public? I am just in the finishing touch of my own .NET lib ...

Have a look at https://github.com/sharpbrick/powered-up

mjorrens, 2020-06-29:
Thanks for checking in. I have not published anything yet but I have a pretty good library now (Visual Studio 2019 - C# - UWP) that handles both the new Lego Technic Control+ and SBrick. I am creating an automated worksite with various lego models. I am using a PixyCam on the ceiling that connects to my laptop via Bluetooth and color codes on each lego model that give me x,y position and orientation (0 to 359 degrees). My Liebherr 9800 moves to an area of color ping pong balls, scoops them up and then a Lego A25F dump truck pulls up alongside and the 9800 drops in the ping pong balls. All this is automated. I am working on the ping pong ball scoop process right now.

Anyway, I do want to get things published but code certainly needs some polishing.

One problem I have come across running my solution is when I connect to both Liebherr Hubs AND request port notifications from all of the motors for a message every 10 rotations (have tried 100 rotations and 1000 as well), the first hub registered begins to be delayed in all motor commands. I have found a workaround by simply doing a Port Information Request to get the current position when I need it. The problem seems to be a Microsoft Bluetooth stack issue. If I run my EXE twice on my machine with one EXE controlling Hub A and the other controlling Hub B, I have the same problem. I do not have the problem if I run one EXE on Computer A and the other EXE on Computer B. That tells me the actual Lego Hub is fine and doing its job.

Not sure if you have run into that or can replicate it. The easy way to test is to register port notification events for all 7 motors (both hubs) and then try to send motor commands to the hub that was first registered. The motor response will become delayed by 1-3 seconds.

Anyway, other than that, I am quite pleased with the new Lego Technic Control+ Hubs.