nathankellenicki/node-poweredup #69: Is there a way to save the absolute position reference point on the Control+ Hub?

mjorrens, 2019-12-21:
I am working on a calibration routine for the Control+ Hub and when I reset the absolute position, I would like that to "stick" between power cycles of the Hub. Not sure that can be saved in the hub or whether I should handle it in my application.

Currently, when the Control+ Hub is powered up, the current position of the motor becomes the reference point (absolute position zero).

aileo, 2019-12-21:
I think the control+ app keeps the position because I have to recalibrate when I used the 42000 with different devices.

I think it uses [this command](https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#output-sub-command-presetencoder-position-n-a) to reset position on power up.

nathankellenicki, 2019-12-22:
I'm afraid not, the hub does not keep the position between power cycles. It does however stick between disconnects and reconnects, as long as the hub hasn't fully powered off.

Technically, there is no hardware difference between the system Linear Motors and the Technic Linear motors, they have no set absolute position, as you've discovered. The only difference is for the Technic motors they allow the hubs internal knowledge of where 0 is to be reset.

You will either need to keep track in your own software, or ensure you always reset the motor to 0 after moving (servo style) to ensure 0 is in the correct location after starting.

My belief is the new Spike Prime motors do have a hardware set absolute zero if you want to wait for those (Lego refer to them as "Technic Angular Motors" instead of "Technic Linear Motors" as they do the new Control+ ones)

mjorrens, 2019-12-23:
Thanks Leo and Nathan. 

Leo's reference to the Lego documentation looks like it will work fine for me and thus allow me to set the position reference point on power-up.

closed 2019-12-23 by mjorrens