nathankellenicki/node-poweredup #70: Help: Won't connect to raspberry pi 3

mylogon341, 2020-01-14:
I have the simple example script work totally fine on my mac, but when I try on my rpi, it just sits on `Scanning for Hubs...`.  
When I run `sudo hcitool lescan` it surely enough finds `90:84:2B:17:8A:5B Smart Hub`, so I'm kind of puzzled. Can anyone help point me in the right direction?  
Thanks

nathankellenicki, 2020-01-17:
Would you be able to try again but setting `export DEBUG=*` on so we can see some logs to help debug?

Mikou, 2020-03-05:
I'm on Ubuntu and have the same kind of problem. In my case, according to the log, it seems to be a permission problem. Here is the complete log:

(Running the JS script with `node test.js`)
```
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +1ms
  hci onSocketError: EPERM, Operation not permitted +1ms
  hci set le event mask - writing: 010120081f00000000000000 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read local version - writing: 01011000 +0ms
  hci write LE host supported - writing: 016d0c020100 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read LE host supported - writing: 016c0c00 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0c0101100006000e060200000e +1ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4097 +0ms
  hci 		status = 0 +0ms
  hci 		result = 06000e060200000e +0ms
  hci set scan enabled - writing: 010c20020001 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci set scan parameters - writing: 010b200701100010000000 +1ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci onSocketData: 040e0a0109100091dd4493fcd8 +0ms
  hci 	event type = 4 +0ms
  hci 	sub event type = 14 +0ms
  hci 		cmd = 4105 +0ms
  hci 		status = 0 +0ms
  hci 		result = 91dd4493fcd8 +0ms
  hci address = d8:fc:93:44:dd:91 +0ms
  noble addressChange d8:fc:93:44:dd:91 +0ms
```

(Running the JS script with `sudo node test.js`)
```
Scanning for Hubs...
Discovered Train Base        !
noble warning: unknown peripheral 806fb0e0b833
```

nathankellenicki, 2020-03-09:
Yeah, under most circumstances this will require elevated privileges (ie. sudo) as it requires access to Bluetooth hardware.

Aside from the warning, it looks like it discovered the Train Base...does it work for you once elevated?

Mikou, 2020-03-15:
When running with sudo I get:

noble warning: unknown peripheral 806fb0e0b833

nathankellenicki, 2020-05-23:
What does your code look like? It looks like it discovers the hub ok, but would need to know what your code tries to do next.

nathankellenicki, 2020-06-12:
Closing due to inactivity.

closed 2020-06-12 by nathankellenicki

mylogon341, 2020-06-21:
Sorry for such a late response. Just to confirm that I did come back to this and running with `sudo` does indeed solve it. Not that I scoured the docs, but is that documented anywhere or did I miss something obvious?
Thanks

aileo, 2020-06-22:
Elevation is not documented in this repo but in the underlaying bluetooth library : noble.

You can find more information here: https://github.com/noble/noble#running-on-linux (and how to run it without sudo)

@nathankellenicki , do you want a PR to add this to the "Known Issues and Limitations" part of the README ?