nathankellenicki/node-poweredup #71: Failing to get Duplo Train working

gnbl, 2020-01-23:
I'm trying this on RPi Zero W  with 

> Raspbian Buster Lite
> Minimal image based on Debian Buster
> Version: September 2019
> Release date: 2019-09-26
> Kernel version: 4.19

(upgraded)

I had issues with nodejs and nmp earlier, so 

```
pi@raspberrypi:~ $ wget https://nodejs.org/dist/latest-v11.x/node-v11.15.0-linux-armv6l.tar.gz
pi@raspberrypi:~ $ tar -xzf node-v11.15.0-linux-armv6l.tar.gz
pi@raspberrypi:~ $ cd node-v11.15.0-linux-armv6l/
pi@raspberrypi:~/node-v11.15.0-linux-armv6l $ sudo cp -R * /usr/
pi@raspberrypi:~/node-v11.15.0-linux-armv6l $ node -v
v11.15.0
pi@raspberrypi:~/node-v11.15.0-linux-armv6l $ npm -v
6.7.0
```

    pi@raspberrypi:~ $ npm install node-poweredup

due to this issue

```
internal/modules/cjs/loader.js:670
    throw err;
    ^

Error: Cannot find module '/home/pi/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:668:15)
    at Function.Module._load (internal/modules/cjs/loader.js:591:27)
    at Module.require (internal/modules/cjs/loader.js:723:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/home/pi/node_modules/@abandonware/bluetooth-hci-socket/lib/native.js:6:15)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Module.require (internal/modules/cjs/loader.js:723:19)

```
I installed this manually:

    pi@raspberrypi:~ $ npm install @abandonware/bluetooth-hci-socket

Using the modified (2nd line -> "DuploTrainBase()") example:

```
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.DuploTrainBase();

poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    console.log(`Discovered ${hub.name}!`);

    await hub.connect(); // Connect to the Hub
    console.log("Connected");

    await hub.sleep(3000); // Sleep for 3 seconds before starting

    while (true) { // Repeat indefinitely
        console.log("Running motor B at speed 75");
        hub.setMotorSpeed("B", 75); // Start a motor attached to port B to run a 3/4 speed (75) indefinitely
        console.log("Running motor A at speed 100 for 2 seconds");
        await hub.setMotorSpeed("A", 100,  2000); // Run a motor attached to port A for 2 seconds at maximum speed (100) then stop
        await hub.sleep(1000); // Do nothing for 1 second
        console.log("Running motor A at speed -50 for 1 seconds");
        await hub.setMotorSpeed("A", -50,  1000); // Run a motor attached to port A for 1 second at 1/2 speed in reverse (-50) then stop
        await hub.sleep(1000); // Do nothing for 1 second
    }
});

poweredUP.scan(); // Start scanning for Hubs
console.log("Scanning for Hubs...");

```

I receive this error:

```
pi@raspberrypi:~ $ export DEBUG=*
pi@raspberrypi:~ $ node nodetest.js
/home/pi/node_modules/node-poweredup/dist/node/hub.js:38
        device.on("disconnect", () => {
               ^

TypeError: Cannot read property 'on' of undefined
    at new Hub (/home/pi/node_modules/node-poweredup/dist/node/hub.js:38:16)
    at new LPF2Hub (/home/pi/node_modules/node-poweredup/dist/node/lpf2hub.js:23:9)
    at new DuploTrainBase (/home/pi/node_modules/node-poweredup/dist/node/duplotrainbase.js:23:9)
    at Object.<anonymous> (/home/pi/nodetest.js:2:19)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11

```

gnbl, 2020-01-23:
On second thought (after looking at https://nathankellenicki.github.io/node-poweredup/DuploTrainBase.html) I reverted the sample and ran it with sudo:

```
pi@raspberrypi:~ $ sudo node nodetest.js
Scanning for Hubs...
Discovered Train Base        !
Connected
Running motor B at speed 75
(node:7476) UnhandledPromiseRejectionWarning: Error: Port B does not exist on this Hub type
    at DuploTrainBase._portLookup (/home/pi/node_modules/node-poweredup/dist/node/hub.js:328:19)
    at DuploTrainBase.setMotorSpeed (/home/pi/node_modules/node-poweredup/dist/node/duplotrainbase.js:61:30)
    at PoweredUP.poweredUP.on (/home/pi/nodetest.js:12:13)
(node:7476) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:7476) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Discovered Train Base        !

```

gnbl, 2020-01-23:
"B" or "AB" are not working either.

gnbl, 2020-01-23:
```
DuploTrainBase {
  _events:
   [Object: null prototype] { connect: [Function], disconnect: [Function] },
  _eventsCount: 2,
  _maxListeners: undefined,
  autoSubscribe: true,
  useSpeedMap: true,
  type: 5,
  _ports:
   { MOTOR:
      Port {
        connected: false,
        busy: false,
        finished: null,
        _eventTimer: null,
        id: 'MOTOR',
        value: 0,
        type: 0 },
     COLOR:
      Port {
        connected: false,
        busy: false,
        finished: null,
        _eventTimer: null,
        id: 'COLOR',
        value: 18,
        type: 0 },
     SPEEDOMETER:
      Port {
        connected: false,
        busy: false,
        finished: null,
        _eventTimer: null,
        id: 'SPEEDOMETER',
        value: 19,
        type: 0 } },
  _virtualPorts: {},
  _name: '',
  _firmwareVersion: '0.0.00.0000',
  _hardwareVersion: '0.0.00.0000',
  _primaryMACAddress: '00:00:00:00:00:00',
  _batteryLevel: 100,
  _voltage: 0,
  _current: 0,
  _rssi: -60,
  _isConnecting: true,
  _isConnected: false,
  _bleDevice:
   NobleDevice {
     _events:
      [Object: null prototype] { disconnect: [Function], discoverComplete: [Function] },
     _eventsCount: 2,
     _maxListeners: undefined,
     _name: 'Train Base\u0000        ',
     _listeners: {},
     _characteristics: { '000016241212efde1623785feabcd123': [Characteristic] },
     _queue: Promise { undefined },
     _mailbox: [],
     _connected: true,
     _connecting: false,
     _noblePeripheral:
      Peripheral {
        _noble: [Noble],
        id: 'xxx',
        uuid: 'xxx',
        address: 'xxx',
        addressType: 'public',
        connectable: true,
        advertisement: [Object],
        rssi: -48,
        services: [Array],
        state: 'connected',
        _events: [Object],
        _eventsCount: 1 },
     _uuid: 'xxx' },
  _ledPort: 17,
  _voltageMaxV: 6.4,
  _voltageMaxRaw: 3047,
  _currentMaxMA: 2444,
  _currentMaxRaw: 4095,
  _lastTiltX: 0,
  _lastTiltY: 0,
  _lastTiltZ: 0,
  _messageBuffer: <Buffer >,
  _voltagePort: 20 }

```

So 

```
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    console.log(`Discovered ${hub.name}!`);
    await hub.connect(); // Connect to the Hub
    console.log("Connected");

//    console.log( hub );
	
    await hub.sleep(3000); // Sleep for 3 seconds before starting

    while (true) { // Repeat indefinitely

        console.log("Running motor MOTOR at speed 75");
        hub.setMotorSpeed("MOTOR", 75); // Start a motor attached to port B to run a 3/4 speed (75) indefinitely
        console.log("Running motor MOTOR at speed 100 for 2 seconds");
        await hub.setMotorSpeed("MOTOR", 100,  2000); // Run a motor attached to port A for 2 seconds at maximum speed (100) then stop
        await hub.sleep(1000); // Do nothing for 1 second
        console.log("Running motor MOTOR at speed -50 for 1 seconds");
        await hub.setMotorSpeed("MOTOR", -50,  1000); // Run a motor attached to port A for 1 second at 1/2 speed in reverse (-50) then stop
        await hub.sleep(1000); // Do nothing for 1 second
    }
});

poweredUP.scan(); // Start scanning for Hubs
console.log("Scanning for Hubs...");

```

works.

gnbl, 2020-01-23:
Sensor test example only receives color sensor (no speedometer, motor does not start when turning front wheels):
```

Looking for Powered UP Hubs...
Connected to Train Base        !
Device attached to port MOTOR (Device ID: 41)
Device attached to port COLOR (Device ID: 43)
Device attached to port SPEEDOMETER (Device ID: 44)
Button press detected (Button: GREEN, State: 1)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 0)
Color detected on port COLOR (Color: 0)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 0)
Color detected on port COLOR (Color: 0)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 0)
Color detected on port COLOR (Color: 9)
Color detected on port COLOR (Color: 9)
Disconnected Train Base

```

nathankellenicki, 2020-01-23:
Yes, the motor port is called "MOTOR" on the Duplo train.

The sensor test example doesn't actually print out any speedometer readings. If you want to add it, you can add:

```js
hub.on("speed", (port, speed) => {
    console.log(`Speed on port ${port} (Speed: ${speed})`);
});
```

I'm not sure why your motor doesn't start - that's not actually part of this library, that's built into the hub itself. It's designed to start moving the motor when the front wheels start turning.

gnbl, 2020-01-24:
Thanks for getting back to me.
Sorry, I mistook the "Device attached to port SPEEDOMETER (Device ID: 44)" for output handled in the sample code - I'll try your suggestion.

Regarding the normal operation of the duplo train base (HUB V.5) - I think it's disabled by the library, as with the original LEGO app and [BrickNil](https://virantha.github.io/bricknil/) (Python) it's not suppressed.

Unfortunately I'm not familiar with nodejs so I won't be digging into this much further, but if I notice something, I'll let you know. Thanks for sharing!

Feel free to close.

aileo, 2020-01-30:
> I'm not sure why your motor doesn't start - that's not actually part of this library, that's built into the hub itself. It's designed to start moving the motor when the front wheels start turning. Either it's designed to not happen when there's a connected device, or maybe something in this library is stopping it from happening? Either way, I'm not sure at this point.

I suspect the subscription to some mode to modify this behavior, but I don't own a Duplo train so I can check for available modes.

nathankellenicki, 2020-02-10:
So I finally got access to my DuploTrainBase to try this out. I can reproduce the behaviour, but to @aileo's point, I removed all outbound subscriptions/requests:

```
Looking for Hubs...
  lpf2hub LPF2Hub connecting +0ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 00 01 29 00 01 00 00 00 01 00 00 00> +442ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 01 01 2a 00 01 00 00 00 01 00 00 00> +2ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 11 01 17 00 01 00 00 00 01 00 00 00> +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 12 01 2b 00 00 00 00 01 00 00 00 01> +11ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 13 01 2c 00 01 00 00 00 01 00 00 00> +1ms
  lpf2hub Received Message (LPF2_ALL) <Buffer 0f 00 04 14 01 14 00 01 00 00 00 01 00 00 00> +0ms
  lpf2hub LPF2Hub connected +338ms
Connected to NK_DuploCargo!
```

And it still doesn't work. So I think the opposite is true - there is something that *isn't* being sent that would enable this functionality.

I'm not sure what BrickNil is doing in this regard but I'll have a look through their code.

nathankellenicki, 2020-02-10:
@gnbl I'd like to confirm the functionality you're _not_ seeing.

If I `setPower` on the Duplo Train motor, it works, and moves the train. Then if I lift the train off the ground, the motor stops, as the front wheels are no longer rotating.

What _doesn't_ work is the automatic starting of the train when pushed with my hand. Is this true with BrickNil too? I can't see any code in BrickNil's repo that specifically enables this.

As for the Lego app, its entirely possible that if this functionality works with that, they're either sending a special command, or subscripting to speed sensor notifications in order to start the motor moving. However we won't know this.

This library can allow you to replicate that functionality if you wish though.

gnbl, 2020-02-12:
Thanks for having another look. It might be useful to use the library just for logging train data during regular use. I was mostly interested in the speed and color readings.
I've just now noticed that the "Train Base" makes starting (horn) and stopping (screeching) sounds when moved manually only when connected to the app - supporting that the app in fact controls the entire train behavior when connected.

nathankellenicki, 2020-02-12:
@gnbl Thanks for confirming!

I reached out to one of my contacts in the Lego firmware team last night who told me that he thinks push-to-go is disabled when a connection is active. He also suggested the solution of reading the speedometer and setting motor speed yourself if you want to replicate the functionality.

My guess is that this is to allow the connected device to have finer grained control over how the train behaves, but obviously that's just a guess. As for what you remarked above, if you choose to revisit this, speed and color sensors should be available to you. :)

I'll close this as resolved, feel free to re-open if you have any further comments.

closed 2020-02-12 by nathankellenicki