nathankellenicki/node-poweredup #72: Spike Prime DeviceTypes

cpseager, 2020-02-06:
For info, the new external sensors/motors may be of interest:

//existing internal ids also used in Spike
VOLTAGE = 20,
CURRENT = 21,
RGB_LIGHT = 23,
CONTROL_PLUS_ACCELEROMETER = 57,
CONTROL_PLUS_GYRO = 58,
CONTROL_PLUS_TILT = 59,

//new Spike internal ids
SPIKE_LED_MATRIX = 25;
RGB_SENSOR = 64; 

//new external devices
SPIKE_MEDIUM_MOTOR = 48,
SPIKE_LARGE_MOTOR = 49,
SPIKE_COLOR = 61;
SPIKE_DISTANCE = 62;
SPIKE_FORCE = 63;

//not released but mentioned often within spike app
SPIKE_SMALL_MOTOR = 65; 

The new devices are all well documented here:
https://education.lego.com/en-us/support/spike-prime/technical-specifications?fbclid=IwAR2ZrltttwUkGmztKrA5RVGJr99Z80A_OsZ8bV6bruQSQMNh3nRk4HVLUSA

nathankellenicki, 2020-02-06:
Hi @cpseager, thanks for this! I actually have support for the Spike Prime sensors/motors in a seperate branch I've yet to publish (as it actually changes the entire way this library works). See here:

https://github.com/nathankellenicki/node-poweredup/blob/feature/devices/src/devices/technicforcesensor.ts
https://github.com/nathankellenicki/node-poweredup/blob/feature/devices/src/devices/technicdistancesensor.ts
https://github.com/nathankellenicki/node-poweredup/blob/feature/devices/src/devices/techniccolorsensor.ts
https://github.com/nathankellenicki/node-poweredup/blob/feature/devices/src/devices/techniclargeangularmotor.ts
https://github.com/nathankellenicki/node-poweredup/blob/feature/devices/src/devices/technicmediumangularmotor.ts

I have a question regarding the internal ports you mentioned. I was under the impression that the Spike Prime hub doesn't support incoming BLE connections or the LWP3 protocol. I've only managed to connect to the hub via serial tty (Bluetooth or USB) to which high-level MicroPython code can be written.

Are those devices exposed any other way? I feel since this library can't support the Spike Prime hub without LWP3 support (or adding support for a different communication method/protocol), there isn't a way to take advantage of them without node-poweredup?

Though I hope to hear different! :)

cpseager, 2020-02-07:
Yes, you are correct AFAIK. You have to connect to spike via bluetooth serial, but then spike can connect to (and control) LWP3 hubs by BLE e.g. using python code like this:

https://github.com/nutki/spike-tools/blob/master/hub/lwp.py

So the HubLEDMatrix is not much use for node-poweredup, but they do appear in the spike app javascript in various places like this, so I assume the app is accessing them with the usual LWP3 ID somehow over serial: 

            function(A) {
                A[A.HubVoltageSense = 20] = "HubVoltageSense", 
		A[A.HubCurrentSense = 21] = "HubCurrentSense", 
		A[A.HubUILight = 23] = "HubUILight", 
		A[A.HubLEDMatrix = 25] = "HubLEDMatrix", 
		A[A.WedoDetect = 35] = "WedoDetect", 
		A[A.BoostVisionSensor = 37] = "BoostVisionSensor", 
		A[A.BoostExternalEncodedMotor = 38] = "BoostExternalEncodedMotor", 
		A[A.BoostInternalEncodedMotor = 39] = "BoostInternalEncodedMotor", 
		A[A.BoostInternalTiltSensor = 40] = "BoostInternalTiltSensor", 
		A[A.ConvoyLMotor = 46] = "ConvoyLMotor", 
		A[A.ConvoyXLMotor = 47] = "ConvoyXLMotor", 
		A[A.FlipperMediumMotor = 48] = "FlipperMediumMotor", 
		A[A.FlipperLargeMotor = 49] = "FlipperLargeMotor", 
		A[A.ThreeAxisAcceleration = 57] = "ThreeAxisAcceleration", 
		A[A.ThreeAxisGyro = 58] = "ThreeAxisGyro", 
		A[A.ThreeAxisOrientation = 59] = "ThreeAxisOrientation", 
		A[A.TemperatureSensor = 60] = "TemperatureSensor", 
		A[A.FlipperColor = 61] = "FlipperColor", 
		A[A.FlipperDistance = 62] = "FlipperDistance", 
		A[A.FlipperForce = 63] = "FlipperForce", 
		A[A.RGBSensor = 64] = "RGBSensor", 
		A[A.FlipperSmallMotor = 65] = "FlipperSmallMotor"
		}
		I = e.LPF2Sensor || (e.LPF2Sensor = {})), 
		e.AllMotorTypes = [I.BoostInternalEncodedMotor, I.BoostExternalEncodedMotor, I.FlipperSmallMotor, I.FlipperMediumMotor, I.FlipperLargeMotor], 
		e.UNKNOWN_MODE = {
                name: "hardware.sensors.names.unknown",
                id: -1,
                default: 0
            };

nathankellenicki, 2020-02-07:
Gotcha. I had seen @nutki's code, it's great work! I hope to use it myself for a project sometime soon. :D

It seems that that snippet comes from some kind of decompiled version of the Spike Prime app, do you have more info on that effort? Would be great to poke around!

I already notice a couple of interesting things from the device types you posted. It looks like they support in some fashion every physical device available right now (except the WeDo tilt sensor), which is great, but there's two new ones there we didn't know about:

```
A[A.RGBSensor = 64] = "RGBSensor", 
A[A.FlipperSmallMotor = 65] = "FlipperSmallMotor"
```

"Flipper" seems to be the name for Spike Prime devices based on that list (`FlipperForce`, `FlipperDistance`, etc), so I assume `FlipperSmallMotor` is a new smaller motor that hasn't been released yet. Also interestingly "Convoy" seems to be the name for Control+ devices.

Similarly, `RGBSensor` is unknown by me at this point, I wonder what it refers to.

cpseager, 2020-02-08:
I’ve done a fair bit of Blockly/Scratch work in the past and was curious whether I could inject my own custom blocks into the app. The spike app is distributed via electron, but uses a V8 heap snapshot for most of the code. This snapshot reduces startup times considerably, but unfortunately also hides most of the code away from nosy people...

The code that isn't contained in the heap snapshot is compressed into app.asar, so that can be extracted and viewed by simply using

     asar extract app.asar destfolder 

The code appears to support spike (for the spike app), EV3 (for the EV3 classroom app) and Boost (?, future LWP3 app?).

And yes, even the firmware file is still called flipper_xxx.bin, so flipper was the internal codename before the marketing people decided on Spike Prime!

My guess is that RGBSensor is the Spike left/right button switches around the RGB LED (like the PUP remote has device 0x37 for the switches).

nathankellenicki, 2020-02-10:
Really interesting, thanks for the info! I've started poking around the code too. Interesting that in their own package.json they've licensed the code as MIT!

nathankellenicki, 2020-02-10:
Closing as `v6.0.0` adds support for the new SPIKE Prime sensors.

closed 2020-02-10 by nathankellenicki