nathankellenicki/node-poweredup #73: Duplo Train Base Speaker - way to play something besides built in sounds?

Taytay, 2020-02-23:
First of all, your library is fantastic. I got the Duplo Cargo Train for my kid, and I'm not sure which of us is having more fun. Thank you for the time you put into this, and for doing it in Typescript as well. That's made it much easier to figure out how to use it.

I've managed to train motor, and I can get the color data from the sensor, and I can play built in sounds. However, I couldn't tell if there is a way to send raw data to the speaker in such a way as to play anything else? Do you know if there is such a way, or know where I would look to find out what sort of data to send in a direct stream?

Many thanks again!

nathankellenicki, 2020-02-24:
Hi, and many thanks for the kind words! I'm glad you're finding this library useful. :)

The Duplo train base doesn't support raw data to be piped to it I'm afraid. There is another mode that seems to allow to specify playing a limited selection of tones (Seems to be 10) but I haven't investigated it yet. Another item for the to-do list!

Taytay, 2020-02-25:
Ah, the tones would actually be perfect (as i was trying to make a simple piano train that drive over construction paper...)

I am happy to experiment with this myself if you have any pointers for putting the hardware in a different mode. I haven’t wrapped my head around the various modes and haven’t seen the code that would allow switching modes for a particular device. Is there any code in the library that illustrates alternative modes?

nathankellenicki, 2020-02-25:
So I just added support for this functionality in v6.1.0, however I feel you may be slightly disappointed. In my testing it only seems to play four "tones" or so - 1-4, and they appear to be system sounds of some kind. ie. bleeps and bloops. 6 and 8 appear to play the Bluetooth connect and disconnect sounds.

If you're interested in how modes work in this library, the functionality is greatly abstracted out. Whenever you add an event handler for a supported input mode, ie "color", the associated mode id is looked up on demand, and the appropriate subscribe command is sent over BLE. For output modes, the subscribe command is sent first before the appropriate command.

Relevant code snippets:

Event name to mode lookup for event handlers: https://github.com/nathankellenicki/node-poweredup/blob/master/src/devices/duplotrainbasecolorsensor.ts#L68-L78
Auto-subscribe functionality whenever an event listener is added: https://github.com/nathankellenicki/node-poweredup/blob/master/src/devices/device.ts#L39-L65
Putting the speaker into sound mode and playing a sound: https://github.com/nathankellenicki/node-poweredup/blob/master/src/devices/duplotrainbasespeaker.ts#L23-L29

Taytay, 2020-02-25:
Thanks for both the code addition for the tones, and the examples in code! Much appreciated. I've learned quite a bit about this ecosystem from your work here and your posts in the Eurobricks forums.

Taytay, 2020-02-25:
I'll go ahead and close this, as my question has been thoroughly answered.

closed 2020-02-25 by Taytay