nathankellenicki/node-poweredup #74: ColorAndDistance example for Duplo train

Taytay, 2020-02-24:
I understand that the duplo train hub can emit both color AND distance information, but so far all I’ve managed to see is the “color” event. If I try instead to subscribe to “colorAndDistance” on the hub then it never seems to fire. From my cursory looking at the code, the Duplo color sensor only seems to operate in one mode and only seems to edit the “color” event. It’s likely that I’m missing something obvious since I’m quite new to this library/ecosystem.

nathankellenicki, 2020-02-24:
Unfortunately the Duplo train base color sensor doesn't support distance. This library used to support "colorAndDistance" on the Duplo Train Base, but it was confusing - as devices weren't separated from hubs and you generically subscribed to events, a small hack was that "colorAndDistance" was fired but without the distance value. With the seperation of devices, only the "color" event is supported on the Duplo train base now.

However you have reminded me that there are other values that I haven't implemented yet - namely RGB values, and reflectivity. I'll add those to my to-do list. :)

Taytay, 2020-02-25:
Oh gotcha. Thank you for the reply. 
To make sure i understand, this is a hardware limitation and not a software one, right? I saw a post (from you i thought)  about duplo train base supporting both color and distance, presumably to help it not get false reads. It makes sense if it’s only color, but i wanted to verify. 

And adding the reflectivity and RGB values sound great. Thanks!

nathankellenicki, 2020-02-25:
Yeah - it was a while back, and I think there was confusion even from myself whereby I _thought_ it supported distance. However on further testing it doesn't appear to. I think the confusion arose due to the implementation mentioned above. Apologies for the incorrect information. :)

Lego LPF2 hubs and devices (for which the Duplo train base is one) have the capability to report back on their capabilities, sensor types, as well as input and output values. For example [here is the Duplo train base and sensors](https://gist.github.com/nathankellenicki/482b97247892032101b3190b893af987), parsed out into a human readable format. As you can see the COLOR sensor only returns a single 8-bit value between 0 and 10, which are the different color indexes.

In other news, I added rgb value and reflectivity in v6.1.0, which I've just published. You can watch for "rgb" or "reflect" events from the Color sensor.

Taytay, 2020-02-25:
Oh, that's helpful. I look forward to playing with those other events too! Thank you!

closed 2020-02-25 by Taytay