nathankellenicki/node-poweredup #75: Unable to connect multiple Hubs on Raspberry Pi 3 B v2

highflying, 2020-03-01:
I am trying to connect multiple hubs on a Raspberry Pi 3 B v2. I'm using a cut down sample script to debug with no success:

```
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();
poweredUP.on("discover", async (hub) => {
    console.log(`Discovered ${hub.name}!`);
    await hub.connect();
    console.log(`Connected to ${hub.name}`);
});
poweredUP.scan();
console.log("Scanning for Hubs...");
```

The first device connects, but no more are detected:

```
root@rasp1:/home/pi/stash/lego-layout/apps/state# node ./dist/test.js
Scanning for Hubs...
Discovered Storage!
Connected to Storage
```

If I run the same thing on my Macbook I can connect to multiple devices:

```
$ node ./dist/test.js 
Scanning for Hubs...
Discovered Storage!
Discovered Tunnel Approac!
Connected to Storage
Connected to Tunnel Approac
Discovered Station Depart!
Discovered Station Approa!
Connected to Station Depart
Discovered Station!
Connected to Station Approa
Connected to Station
```

I have double checked that I have installed the correct dependencies on the Raspberry Pi.

I'm guessing this is a problem more with `noble` rather than `node-poweredup` - any suggestions?

aileo, 2020-03-01:
Hi, I had a similar problem on my Ubuntu 16.04. I think you have to make another call to `poweredUP.scan();` in the discover handler because it stops after every discover.

highflying, 2020-03-01:
I have since got it working by calling scan again after connecting:

```
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();
poweredUP.on("discover", async (hub) => {
    console.log(`Discovered ${hub.name}!`);
    await hub.connect();
    console.log(`Connected to ${hub.name}`);
    poweredUP.scan();
});
poweredUP.scan();
console.log("Scanning for Hubs...");
```

Is this a bug or is this what I should be doing?

highflying, 2020-03-01:
Now that I have 5 Technic Control+ Hubs and 2 regular PoweredUp Hubs connected at the same time, I have found that that not everything works that well. I have had to put a semaphore around all the poweredup calls and now everything runs much faster, all 7 hubs can be switched on at the same time and connect ok, other commands seem to be better now too.

GianCann, 2020-03-02:
@highflying can you confirm 7 device connected simultaneously to Raspy 3B+?

highflying, 2020-03-02:
Yes:

[image]

highflying, 2020-03-02:
> Hi, I had a similar problem on my Ubuntu 16.04. I think you have to make another call to `poweredUP.scan();` in the discover handler because it stops after every discover.

Strange however that it isn't needed when running on my Macbook.

aileo, 2020-03-02:
I think it is similar to https://github.com/nathankellenicki/node-poweredup/issues/59#issuecomment-561423756

Noble seems to handle things differently between mac and unix systems.

highflying, 2020-03-02:
> @highflying can you confirm 7 device connected simultaneously to Raspy 3B+?

I've just connected my PoweredUp Remote, so that's now 8 devices connected at the same time.

nathankellenicki, 2020-03-02:
@highflying Glad you got it working! As you discovered, due to the way noble works with certain Bluetooth adapters, calling scan again is necessary. This library used to do it automatically, but issues were discovered - namely, I couldn't find/dictate a proper place to call it.

You mentioned once you added a semaphore around calls the library worked more reliably - can you go into more detail as to the issues you were encountering? Woud be good to know if there's something I can improve there.

highflying, 2020-03-02:
> You mentioned once you added a semaphore around calls the library worked more reliably - can you go into more detail as to the issues you were encountering? Woud be good to know if there's something I can improve there.

There were a couple of things that made me try using a semaphore (which I have stuck with using).

- When trying to connect multiple hubs at the same time quite often they wouldn't all connect (although they still don't always connect properly, the Technic Control hubs seem to be better at it, it's the normal PoweredUp hubs that I tend to have more trouble with)

- I was using the trainMotor.rampPower() function but that didn't seem to work so well at times (not always setting the correct speed). Although the semaphore hasn't helped with this, perhaps there's something about controlling 2 different train motors on 2 different hubs at the same time?

Overall, things do seem a lot more stable by using a semaphore to ensure only 1 command is issued at a time, for instance all the hubs connect a lot faster this way.

highflying, 2020-03-03:
> @highflying can you confirm 7 device connected simultaneously to Raspy 3B+?

Mine is a 3B not a 3B+

nathankellenicki, 2020-03-03:
> There were a couple of things that made me try using a semaphore (which I have stuck with using).
>
> * When trying to connect multiple hubs at the same time quite often they wouldn't all connect [...]

Got it, thanks! Those two things certainly seem like things that might cause issues with multiple hubs on a Raspberry Pi. Particularly, the race conditions with connecting to a hub then rescanning.

The rampMotorSpeed one I can see too depending on how the BLE stack works on the Pi. There is no build in functionality for ramping speed on train motors, so this library adds it by repeatedly calling setSpeed in a constantly increasing fashion. This could cause these calls to be backed up or lost if calls too fast.

I'll add something to my todo list to see about improving the situation, particularly for the multiple hubs connecting on the Pi.

highflying, 2020-03-03:
I've been doing some more digging today, and it seems that my problems may be coming from the bluetooth stack on Macbook, so I switched to doing more testing on my Pi3, and that was coming up with some other problems! I've found out that the Pi3 doesn't have any flow control on it's inbuilt Bluetooth modem, so that could explain dropped messages.

I've tried a USB bluetooth dongle on the Pi3 and that does seem to be working better, however I've ordered a Pi4 so fingers crossed that'll sort out my issues.

I'll close this issue, but that are some bits that I have come across that I'll raise some PRs with some suggestions for you to take a look at.

closed 2020-03-03 by highflying

highflying, 2020-03-04:
Just wanted to note, now I'm using a Raspberry Pi 4 everything is running a lot better. Definitely seems like the Pi3 couldn't handle it!

GianCann, 2020-03-04:
> now I'm using a Raspberry Pi 4 everything is running a lot better.

It's time for me to buy a Pi4 ;)