nathankellenicki/node-poweredup #78: jetson nano, ubuntu 18, npm install

dftpnd, 2020-03-15:
npm install node-poweredup --save

> usb@1.6.2 install /home/bb8/www/controller/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.3
prebuild-install info looking for cached prebuild @ /home/bb8/.npm/_prebuilds/31517a-usb-v1.6.2-node-v57-linux-arm64.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.2/usb-v1.6.2-node-v57-linux-arm64.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.2/usb-v1.6.2-node-v57-linux-arm64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=8.17.0 runtime=node arch=arm64 libc= platform=linux)
make: Entering directory '/home/bb8/www/controller/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
../src/node_usb.cc: In function ‘void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)’:
../src/node_usb.cc:151:58: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declaration]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../node_modules/nan/nan.h:1001:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
  CXX(target) Release/obj.target/usb_bindings/src/device.o
../src/device.cc: In static member function ‘static void Req::default_after(uv_work_t*)’:
../src/device.cc:237:64: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
    Nan::MakeCallback(device, Nan::New(baton->callback), 1, argv);
                                                                ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/device.cc:1:
../node_modules/nan/nan.h:959:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
../src/transfer.cc: In function ‘void handleCompletion(Transfer*)’:
../src/transfer.cc:126:72: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
   Nan::MakeCallback(self->handle(), Nan::New(self->v8callback), 3, argv);
                                                                        ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/transfer.cc:1:
../node_modules/nan/nan.h:959:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
  SOLINK_MODULE(target) Release/obj.target/usb_bindings.node
  COPY Release/usb_bindings.node
make: Leaving directory '/home/bb8/www/controller/node_modules/usb/build'

> @abandonware/bluetooth-hci-socket@0.5.3-5 install /home/bb8/www/controller/node_modules/@abandonware/bluetooth-hci-socket
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using needle for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(404): https://github.com/abandonware/node-bluetooth-hci-socket/releases/download/0.5.3-5/binding-0.5.3-5-node-v57-linux-arm64.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for @abandonware/bluetooth-hci-socket@0.5.3-5 and node@8.17.0 (node-v57 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/bb8/www/controller/node_modules/@abandonware/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
  COPY /home/bb8/www/controller/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/bb8/www/controller/node_modules/@abandonware/bluetooth-hci-socket/build'

> @abandonware/noble@1.9.2-5 install /home/bb8/www/controller/node_modules/@abandonware/noble
> node-gyp rebuild

make: Entering directory '/home/bb8/www/controller/node_modules/@abandonware/noble/build'
  SOLINK_MODULE(target) Release/obj.target/noble.node
  COPY Release/noble.node
make: Leaving directory '/home/bb8/www/controller/node_modules/@abandonware/noble/build'
npm WARN controller@1.0.0 No description
npm WARN controller@1.0.0 No repository field.

+ node-poweredup@6.1.0
added 305 packages from 220 contributors and audited 686 packages in 44.431s

4 packages are looking for funding
  run `npm fund` for details

found 1 moderate severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details

dftpnd, 2020-03-15:
bb8@bb8:~/www/controller$ node -v
v8.17.0
bb8@bb8:~/www/controller$ npm -v
6.14.2

nathankellenicki, 2020-05-23:
Hi...can I ask what the actual problem you're having is?

nathankellenicki, 2020-06-12:
Closing due to inactivity.

closed 2020-06-12 by nathankellenicki