nathankellenicki/node-poweredup #79: Support for Power Functions

highflying, 2020-03-22:
Now that Power Functions can be support via use of the color sensor I was thinking about having a go at adding support here - does anyone have any documentation that may help me? Or any ideas on where to start!

Thanks in advance.

highflying, 2020-03-22:
A demo of it working via the Powered Up App - https://www.youtube.com/watch?v=scvifoxEbWs

dlech, 2020-03-22:
This is the protocol: https://philohome.com/pf/LEGO_Power_Functions_RC.pdf

The sensor seems to take care of the checksum, 0 can always be used for that nibble.

nutki, 2020-03-22:
I got it to work a couple of months back with node-poweredup but did not publish the PR in the end. I can try to update it to the latest version.

nutki, 2020-03-22:
@dlech The link to the proper version is: https://www.philohome.com/pf/LEGO_Power_Functions_RC_v120.pdf
It is not easy to find as the one you linked is the top Google answer so I also started with the wrong one which led to some frustration as there are significant changes between the versions.

dlech, 2020-03-22:
Oops, thanks!

aileo, 2020-03-23:
I don't know if it will help: mode `0x07` of the color and distance sensor is named `IR_Tx` so I assume it is used somehow to send data over IR.

I wonder if it have any interest to start with a simple IR serial implementation then add the LEGO power functions layer or going straight to the point. Who wants to control its TV through this lib ?

nutki, 2020-03-24:
@aileo As far as I can tell the `IR_Tx` mode is PF IR protocol specific and while the protocol itself is probably in the hub firmware (so could be updated in the future), I think it is unlikely that we will see generic IR support via the color sensor.

GianCann, 2020-03-24:
> probably

Remove "probably".... ;)

nutki, 2020-03-24:
> The sensor seems to take care of the checksum, 0 can always be used for that nibble.

As far as I remember the firmware also interprets the command code to identify if it is "continuous" or "timeout".  "continuous" commands are emitted once (or single burst) while "timeout" commands will be send continuously until the next code input or sensor mode change.

nathankellenicki, 2020-03-24:
So I've implemented a first pass at the Power Functions IR stuff in this PR: https://github.com/nathankellenicki/node-poweredup/pull/80. Right now it implements "Single output mode" and "Combo PWM mode", as well as a method for sending raw PF IR messages.

I've going to add the extended mode channel switching too. Also unless there's some demand for it I wasn't going to implement "Combo direct mode" - it seems that "Combo PWM mode" can achieve exactly the same functionality, unless I'm missing something?

One thing to note is that this is all I'm untested as all my Power Functions stuff is in another country. :) So I'd love some assistance in testing it.

nutki, 2020-03-24:
@nathankellenicki This is part of my PR to be that does the code calculation: https://gist.github.com/nutki/e5c2987ffb897a25ef3dce21ec27b8b5
It was mostly exploratory and I wrote it in time when the sensor API was redesigned so I did not publish it in the end. It has everything in the spec including stuff from protocol version 1.0 which I found is not implemented by my PF receivers (`pinTimount` and `pinContinuous`, as far as I remember).
I think the two commands that in practice cover the whole functionality are the ones you have already in your PR (that is why I exported them directly in my code as `comboPWM` and `singlePWM`).

nathankellenicki, 2020-04-05:
Thanks @nutki for the pointers!

Apologies it took a while, but I ended up ordering some more Power Functions stuff from Bricklink, as I was uncomfortable publishing this without testing. Turns out there were indeed a couple of bugs. :)

I've just published `v6.2.0` which contains support for the Power Functions protocol.

nathankellenicki, 2020-04-05:
Closing as feature added.

closed 2020-04-05 by nathankellenicki