nathankellenicki/node-poweredup #81: DuploTrainBaseColorSensor

jfedyczak, 2020-04-13:
This class only reports reflectivity. Is there any way to request RGB or color events?

nathankellenicki, 2020-04-13:
Hi, yes, it already supports both RGB and color events - https://github.com/nathankellenicki/node-poweredup/blob/master/src/devices/duplotrainbasecolorsensor.ts

You can add event handlers for "color" or "rgb" events to the sensor to receive them.

jfedyczak, 2020-04-13:
I did exactly that, but only `reflect` event is being emitted.

```
	const colorSensor = await trainHub.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_COLOR_SENSOR);
		colorSensor.on('color', color => {
			console.log('color', color)
		})
		colorSensor.on('rgb', rgb => {
			console.log('rgb', rgb)
		})
```
These two events never fire...

jfedyczak, 2020-04-13:
Reflectivity is being reported correctly though. There's `receive` method in `Device` class that depends on `this._mode`. Only place I see that sets `this._mode` is autosubscribe to the list of three events. This means `this._mode` gets overridden by last value subscribed. Do you think it may be the reason of problems? I also tried to manually unscubscibe reflectivity event but it didn't help...

nathankellenicki, 2020-04-13:
> Reflectivity is being reported correctly though. There's `receive` method in `Device` class that depends on `this._mode`. Only place I see that sets `this._mode` is autosubscribe to the list of three events. This means `this._mode` gets overridden by last value subscribed. Do you think it may be the reason of problems? I had also tried to manually unscubscibe reflectivity event but it didn't help...

That's correct, unfortunately at the moment only one mode can be subscribed to at a time. There is a combined subscription mode documented in the protocol, but I haven't gotten it to universally work yet. I have another branch with an early version but it doesn't work well yet.

For the moment only one at a time unfortunately. :)

closed 2020-04-29 by nathankellenicki