nathankellenicki/node-poweredup #83: Can't control duplo train.

p8nut, 2020-05-16:
Hy Everyone,
I am currently trying to control a duplo train (10875) but it seem that i am stuck when i wait to get the motor.
currently the promise of `waitForDeviceByType` never resolve.
Do you have any idea where that can come from ?

here is the code i use:
```javascript
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

console.log("Looking for Hubs...");

poweredUP.on("discover", async (hub) => {

    await hub.connect();
    if (hub instanceof PoweredUP.DuploTrainBase) {
        console.log('train')
        let train = hub;
        console.log("await Motor")
        let motor = await train.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_MOTOR);
        console.log(motor)
    }
});

poweredUP.scan();
```

thanks for your help.

p8nut, 2020-05-18:
i am able to control the train if i rollback to 5.0.3

nathankellenicki, 2020-05-23:
Hi, apologies for the delay. However I'm unable to reproduce this with your code sample - the motor object is logged almost immediately (6.2.0).

What is your environment? OS, Node/WebBluetooth, etc?

p8nut, 2020-05-24:
Hi, i am on a fedora 30
node v10.19.0
node-poweredup 6.3.0
(i just try again with the 6.2.0 it doesn't work too on my computer.)

nathankellenicki, 2020-05-24:
Could you try running it with `export DEBUG=*` and let me know what the debug log says?

p8nut, 2020-05-25:
Here is the [dump.log](https://github.com/nathankellenicki/node-poweredup/files/4676092/dump.log) of the program.
I add some console.log to my code sample here the code 
```javascript
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

console.log('########## DEBUG => Looking for Hubs...');

poweredUP.on('discover', async (hub) => {
    console.log('########## DEBUG => new Hub discovered:', hub);
    await hub.connect();
    console.log('########## DEBUG => connection done')
    if (hub instanceof PoweredUP.DuploTrainBase) {
        console.log('########## DEBUG => its a train')
        let train = hub;
        console.log('########## DEBUG => await Motor')
        let motor = await train.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_MOTOR);
        console.log('########## DEBUG => await motor done: ',motor)
    }
});

poweredUP.scan();
```

aileo, 2020-05-25:
It's been a while since I had the time to look at the LWP but I think I found the attach messages from the logs :

```
2020-05-25T08:20:41.920Z hci onSocketData: 02012e1600120004001b0b000f0004000129000100000001000000
2020-05-25T08:20:41.920Z hci 	event type = 2
2020-05-25T08:20:41.920Z hci 		cid = 4
2020-05-25T08:20:41.920Z hci 		handle = 3585
2020-05-25T08:20:41.920Z hci 		data = 1b0b000f0004000129000100000001000000
```
`0f 00 04 00 01 29 00 01 00 00 00 01 00 00 00`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x00 (motor)
 - event type 0x01 (attached I/O)
 - device type 0x29 (41, duplo train base motor)
 - 0x00
 - hardware and software versions 0x01000000 twice

```
2020-05-25T08:20:41.935Z hci onSocketData: 02012e1600120004001b0b000f000401012a000100000001000000
2020-05-25T08:20:41.935Z hci 	event type = 2
2020-05-25T08:20:41.935Z hci 		cid = 4
2020-05-25T08:20:41.935Z hci 		handle = 3585
2020-05-25T08:20:41.935Z hci 		data = 1b0b000f000401012a000100000001000000
```
`0f 00 04 01 01 2a 00 01 00 00 00 01 00 00 00`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x01 (not found in port map)
 - event type 0x01 (attached I/O)
 - device type 0x2a (42, duplo train base speaker)
 - hardware and software versions 0x01000000 twice

```
2020-05-25T08:20:41.950Z hci onSocketData: 02012e1600120004001b0b000f0004110117000100000001000000
2020-05-25T08:20:41.950Z hci 	event type = 2
2020-05-25T08:20:41.950Z hci 		cid = 4
2020-05-25T08:20:41.950Z hci 		handle = 3585
2020-05-25T08:20:41.950Z hci 		data = 1b0b000f0004110117000100000001000000
```
`0f 00 04 11 01 17 00 01 00 00 00 01 00 00 00`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x11 (led)
 - event type 0x01 (attached I/O)
 - device type 0x17 (23, RGB LED)
 - hardware and software versions 0x01000000 twice

```
2020-05-25T08:20:41.950Z hci onSocketData: 02012e1600120004001b0b000f000412012b000000000100000001
2020-05-25T08:20:41.950Z hci 	event type = 2
2020-05-25T08:20:41.950Z hci 		cid = 4
2020-05-25T08:20:41.950Z hci 		handle = 3585
2020-05-25T08:20:41.950Z hci 		data = 1b0b000f000412012b000000000100000001
```
`0f 00 04 12 01 2b 00 00 00 00 01 00 00 00 01`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x12 (18, COLOR)
 - event type 0x01 (attached I/O)
 - device type 0x2b (43, duplo train base color)
 - hardware and software versions 0x01000000 twice

```
2020-05-25T08:20:41.951Z hci onSocketData: 02012e1600120004001b0b000f000413012c000100000001000000
2020-05-25T08:20:41.951Z hci 	event type = 2
2020-05-25T08:20:41.951Z hci 		cid = 4
2020-05-25T08:20:41.951Z hci 		handle = 3585
2020-05-25T08:20:41.951Z hci 		data = 1b0b000f000413012c000100000001000000
```
`0f 00 04 13 01 2c 00 01 00 00 00 01 00 00 00`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x13 (19, SPEEDOMETER)
 - event type 0x01 (attached I/O)
 - device type 0x2c (44, duplo train base speedometer)
 - hardware and software versions 0x01000000 twice

```
2020-05-25T08:20:41.952Z hci onSocketData: 02012e1600120004001b0b000f0004140114000100000001000000
2020-05-25T08:20:41.952Z hci 	event type = 2
2020-05-25T08:20:41.952Z hci 		cid = 4
2020-05-25T08:20:41.952Z hci 		handle = 3585
2020-05-25T08:20:41.952Z hci 		data = 1b0b000f0004140114000100000001000000
```
`0f 00 04 14 01 14 00 01 00 00 00 01 00 00 00`
 - length 0x0f (15)
 - hub ID 0x00 (always 0x00)
 - message type 0x04 (Hub attached I/O)
 - port 0x14 (voltage)
 - event type 0x01 (attached I/O)
 - device type 0x14 (20, voltage)
 - hardware and software versions 0x01000000 twice

They are received before `bledevice Service/characteristic discovery started` and none of them is followed by the `lpf2hub Received Message (LPF2_ALL) ...` log so I think they are ignored/lost.
If I remmember correctly, the mailbox is not used on the node side but it seems some messages can be sent by the hub before the lib actually listen to them.
It looks very close to the symptoms of #63 but we did not have all the hci logs to dig in.

Side note : There is always 0X00 between device type and version which I omitted, the LWP3 doc says the device type is Uint16 (like 0x0014 for voltage but we found 0x1400). I don't understand that part.

p8nut, 2020-05-25:
is it not possible to ask the HUB to resend data if the devices list is empty of if the device type is not in the list ?

aileo, 2020-05-25:
Missing devices is a normal state for most of the hubs. Device attachment is an event thing from the protocol point of vue, as far as I understand it.

I did not found a downstream couterpart for [attached I/O message](https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#hub-attached-i-o) to ask for "what is connected there".

If I am correct with the lost messages, I think the best fix is to store/process them. The web BLE version already does this kind of thing.

p8nut, 2020-05-25:
but there is a port information request which could be send id no device is connected to a port in the list

https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#port-information-request

aileo, 2020-05-25:
The port information request won't give you the type of device connected, it will respond with the sensor value, the available modes or mode combinations.
Depending on the hub type, modes may differ for the same device, it would be unreliable to guess the device from this information.

It just hit me that attach messages are sent before `bledevice Service/characteristic discovery started` in your log, and the characteristic discovery is made before subscription where `LPF2Hub._parseMessage` is attached to the characteristic data event.

I guess the hci logs come from `@abandonware/bluetooth-hci-socket` wich is optionnal for `@abandonware/noble`. I wonder if it could be the cause of this behavior.

Note: You can control the train with the 5.0.3 because this version does not wait for the devices, you can send message to a port even if there is nothing connected.

nathankellenicki, 2020-05-25:
> It just hit me that attach messages are sent before `bledevice Service/characteristic discovery started` in your log, and the characteristic discovery is made before subscription where `LPF2Hub._parseMessage` is attached to the characteristic data event.

This is what I see too. It would appear that certain hardware/stack/OS/software combinations seem to attempt delivery of data, even if there's no listener attached to receive it, causing the data to be lost. Ideally there should be some kind of mailbox to hold the message until there is a listener ready, but it doesn't appear Noble has that functionality.

I also don't see anywhere where we could attach a listener fast enough to stash the data in a mailbox until ready - it's already done in the earliest place I can think of (minus a couple of layers of promises, but those should be executed in the same thread tick).

I'll have a dig through Noble to see if there's any way we can listen to messages earlier.

p8nut, 2020-12-13:
yesterday i was able to make it work.
the catch is to stop the scan before connect.

```c
const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

console.log("Looking for Hubs...");

poweredUP.on("discover", async (hub) => {
	await poweredUP.stop();
    await hub.connect();
    if (hub instanceof PoweredUP.DuploTrainBase) {
        console.log('train')
        let train = hub;
        console.log("await Motor")
        let motor = await train.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_MOTOR);
        console.log(motor)
    }
});

poweredUP.scan();
```

closed 2020-12-22 by p8nut