nathankellenicki/node-poweredup #87: Can't move Spike Prime Large Motor

corbin-c, 2020-05-27:
Hi,
First of all, I wanna thank you for your amazing work. It is really great to be able to use Lego Powered Up from node js and not from a Scratch-like interface on a smartphone ! :)

I have a  Move Hub and a Spike Prime Large Motor, which I'm trying to use from a Raspberry Pi 3+ running Raspbian. The Move Hub itself is correctly detected and I've been able to control the A & B motors.

Yet I'm unable to use the Spike Prime Large Motor. It has to be noted that I got it from the [#45680 set](https://brickset.com/sets/45680-1/Expansion-Set) and not from the "main" set. Maybe it has something to do with the problem I'm facing.

I'm trying to run a simple test script:

```
poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    console.log(`Discovered ${hub.name}!`);
    await hub.connect(); // Connect to the Hub
    const motorC = await hub.waitForDeviceAtPort("C");
    console.log("Connected");
    console.log("Going to zero");
    motorA.gotoRealZero();
    console.log("going to 90 deg");
    motorA.gotoAngle(90) 
        .then(e => console.log(e)).catch(e => console.error(e));
    console.log(motorA.mode,motorA.type,motorA.connected,);
    process.exit();
});
```

Here's the output I get (with DEBUG=*) (I spare you the whole connection part, I might include it if you need it):

```
Connected
Going to zero
  lpf2hub Sent Message (LPF2_ALL) <Buffer 0a 00 41 02 03 01 00 00 00 01> +4ms
  lpf2hub Sent Message (LPF2_ALL) <Buffer 05 00 21 02 00> +1ms
going to 90 deg
  lpf2hub Sent Message (LPF2_ALL) <Buffer 0e 00 81 02 11 0d 5a 00 00 00 64 64 7f 03> +4ms
3 49 true  hci set scan enabled - writing: 010c20020001 +42ms
  hci disconnect - writing: 01060403400013 +1ms
```

I hope you can help. Thanks again!

nathankellenicki, 2020-05-27:
Hi @corbin-c - unfortunately I have some bad news for you. There is a well known bug in the firmware of the Move Hub which renders Control+ and SPIKE Prime motors incompatible with it at the moment. Without a firmware update from LEGO, they are unusable, even with the official LEGO Powered UP app.

For the moment Control+ and SPIKE Prime motors will work with the standard 2-port Hub as well as the 4-port Control+ Hub.

nathankellenicki, 2020-05-27:
PS. As an additional note, you'll likely want to call `resetZero()` after the `gotoRealZero()` in your code above:

```js
await motorC.gotoRealZero();
await motorC.resetZero();
```

The reason is that the motors have both a physical and logical zero. Logical zero is set to wherever the motor was when initialized, but physical zero is constant. If you `gotoRealZero()` then `gotoAngle(90)`, then you will go to 90 degrees from logical zero rather than physical zero.

You can use `resetZero()` to reset the logical zero to current position.

corbin-c, 2020-05-27:
Well, thank you very much for this quick answer. I'll guess I have no choice but to wait for an official firmware update... Any idea how often those are released ?

nathankellenicki, 2020-05-27:
It's not very often I'm afraid, this bug has existed for about 3-4 months now. In my experience they tend to time firmware updates with new Powered Up features introduced in a new Lego set.

There are rumors there's a new train motor due out in July, which might trigger a PUP app/firmware update, but who knows.

GianCann, 2020-05-27:
> There are rumors there's a new train motor due out in July, which might trigger a PUP app/firmware update, but who knows.

Really? What type of difference with the current train motor?

nathankellenicki, 2020-05-27:
> > There are rumors there's a new train motor due out in July, which might trigger a PUP app/firmware update, but who knows.
> 
> Really? What type of difference with the current train motor?

It's apparently for a new train to be released in July, and has the part number 88013, but apart from that nothing else seems to be known.

corbin-c, 2020-05-30:
Hi, you said:

> There is a well known bug in the firmware of the Move Hub which renders Control+ and SPIKE Prime motors incompatible with it at the moment.

Any sources I could consult about this well known bug ? Thank you again

nathankellenicki, 2020-05-30:
> Hi, you said:
> 
> 
> 
> > There is a well known bug in the firmware of the Move Hub which renders Control+ and SPIKE Prime motors incompatible with it at the moment.
> 
> 
> 
> Any sources I could consult about this well known bug ? Thank you again

The Facebook group “The LEGO Powered Up Community” has a few discussions on it, some new some old, discussing this. I think you’ll need to join to read them though.

corbin-c, 2020-05-30:
Great, thanks again! I guess we may close this issue now

closed 2020-05-30 by corbin-c