nathankellenicki/node-poweredup #89: Boost with spike motors documentation

cpseager, 2020-06-04:
I think these ones should be no until the Boost firmware issues is fixed, I couldn't get them to work. Force did work ok though,.

[image]

cpseager, 2020-06-06:
And for completeness Spike Prime Force does give a valid reading on Wedo too (so partial).

nathankellenicki, 2020-06-12:
@cpseager Thanks, and good point. It's annoying, as no doubt Lego will release an updated firmware that fixes the bug with no change in this library code, and I'll have to update the documentation again. >.<

Also thanks for the heads up about WeDo. It's been a while since I've done a full pass with all the devices against all hubs. Perhaps I'll try to do that this weekend.

nathankellenicki, 2020-06-12:
Interesting. The force sensor seems to be analog in nature, using simple resistance for its value. It doesn't talk the LPF2 serial protocol as used in other motors/sensors. That would explain why it works. Likely only one mode though.

The only other devices like this are the WeDo 2.0 sensors, and the basic and train motors.

cpseager, 2020-06-12:
All the sensors talk serial, I disassembled a wedo tilt a few years back and it is microcontroller based. Only wedo motor/lights/train motor are dumb.

The difference is that the wedo sensors have a 'default mode' at power up (ie they start giving out readings without any configuration). Force must do the same.

nathankellenicki, 2020-06-12:
Updated in `v6.6.0`. Thanks!

closed 2020-06-12 by nathankellenicki