nathankellenicki/node-poweredup #90: Manually detach device

aileo, 2020-06-04:
The introduction of `manuallyAttachDevice` in e6c6a3c18f5ab015ebf06d66687ea4e15f8c5725 makes me suggest to add the detach version of it.

It could allow to virtually remove a device and prevent messages to be sent and events to be fired. However, this is probably pointless :)

nathankellenicki, 2020-06-05:
Well, the purpose of this function is slightly sneaky - its to enable those people who are having troubles with waiting for devices to be attached to manually attach one if they know what device should be at that port. :)

I wish this was something I could get to the bottom at the moment - I have a Raspberry Pi 4 en route which I hear manifests this issue, so I'll try to figure something out. But until then this function helps get around it.

Can't think of a use for `manuallyRemoveDevice` though. ;) No events would be fired if you don't have any event listeners added...no messages sent unless you call a method. And if the device has actually detached this library would throw an error...

aileo, 2020-06-05:
I agree, useless idea.

closed 2020-06-05 by aileo