nathankellenicki/node-poweredup #91: Disable color distance sensor

RaymondNg2, 2020-06-05:
I have connected a color and distance sensor (88007) on a train (60197).
Because of physical space limitation I cannot create a train track loop. I have made a track with 3 stations and written a script to have the train stop at the stations and reverse at the end station after sleeping.
I have the train sleep for 15min at the end stations and 2 min at the middle station. I have also programmed operational hours between 8:00 and 20:00. This setup is intended as a display to run till the batteries are depleted.

My problem is the battery drainage. In the non operational hours the color and distance sensor stays on and drains the batteries. Is it possible to turn off the color and distance sensor just before sleeping and enable it back after sleep is done?

RaymondNg2, 2020-06-05:
This is the code
```
// Double rail with station and single rail at both ends
// For poweredup v6.4.1

const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

var DirectionEnum = {
  FORWARD: 1,
  BACKWARD: 2,
  
  properties: {
    1: {name: "forward"},
    2: {name: "backward"}
  },
  
	reverse: function(direction) {
		if (direction === DirectionEnum.FORWARD) {
			return DirectionEnum.BACKWARD;
		}
		return DirectionEnum.FORWARD;
	}
};

/**
 * @typedef Color
 * @property {number} BLACK 0
 * @property {number} PINK 1
 * @property {number} PURPLE 2
 * @property {number} BLUE 3
 * @property {number} LIGHT_BLUE 4
 * @property {number} CYAN 5
 * @property {number} GREEN 6
 * @property {number} YELLOW 7
 * @property {number} ORANGE 8
 * @property {number} RED 9
 * @property {number} WHITE 10
 * @property {number} NONE 255
 */

var ColorSpeedEnum = {
	BLACK: {name: "Black", colorId: 0,  speed: 0, sleepMilis: 0, reverse: false},
	PINK: {name: "Pink", colorId: 1,  speed: 0, sleepMilis: 0, reverse: false},
	PURPLE: {name: "Purple", colorId: 2,  speed: 0, sleepMilis: 0, reverse: false},
	BLUE: {name: "Blue", colorId: 3,  speed: 0, sleepMilis: 0, reverse: false},
	LIGHT_BLUE: {name: "Light blue", colorId: 4,  speed: 0, sleepMilis: 0, reverse: false},
	CYAN: {name: "Cyan", colorId: 5, speed: 0, sleepMilis: 0, reverse: false},
	GREEN: {name: "Green", colorId: 6, speed: 0, sleepMilis: 5000, reverse: false},
	YELLOW: {name: "Yellow", colorId: 7,  speed: 0, sleepMilis: 0, reverse: false},
	ORANGE: {name: "Orange", colorId: 8,  speed: 0, sleepMilis: 0, reverse: false},
	RED: {name: "Red", colorId: 9, speed: 0, sleepMilis: 900000, reverse: true},
	WHITE: {name: "White", colorId: 10,  speed: 0, sleepMilis: 60000, reverse: false},
	
	lookupByColorId: function(colorId) {
		//log('got colorId: ' + colorId);
		for (var app in ColorSpeedEnum) {
			if (isNaN(Number(app))) {
				if (ColorSpeedEnum.hasOwnProperty(app) && ColorSpeedEnum[app].colorId === colorId) {
					//log('got color: ' + ColorSpeedEnum[app].name);
					return ColorSpeedEnum[app];
				}
			}
		}
	 }
};

var direction = DirectionEnum.FORWARD;
var moving = false;
var handlingColor = false;
[... 200 lines cut ...]

function getFormattedDate(timeStamp) {
    var date = timeStamp === undefined ? new Date() : new Date(timeStamp);

    var month = date.getMonth() + 1;
    var day = date.getDate();
    var hour = date.getHours();
    var min = date.getMinutes();
    var sec = date.getSeconds();

    month = (month < 10 ? "0" : "") + month;
    day = (day < 10 ? "0" : "") + day;
    hour = (hour < 10 ? "0" : "") + hour;
    min = (min < 10 ? "0" : "") + min;
    sec = (sec < 10 ? "0" : "") + sec;

    return date.getFullYear() + "-" + month + "-" + day + " " +  hour + ":" + min + ":" + sec;
}

function isOperationHours() {
	var date = new Date();
	return date.getHours() >= 8 && date.getHours() < 20;
}

```

aileo, 2020-06-05:
First thing I can think of is to remove listener on color and distance sensor when it is not needed.

You can have an interval to manage the operational hours : add/remove color&distance listener, start/stop train and the color&distance listener would be simplier as it would only manage the actual color and distance related operations.

RaymondNg2, 2020-06-05:
Thanks for the response. I will give it a try tomorrow and report back here.

RaymondNg2, 2020-06-06:
I did some testing with unsubscribing the color and distance sensor. Indeed it stopped getting values from the sensor. I did notice that the light on the sensor keeps burning. I still have to verify how the battery drainage is compare to the situation when I did not unsubscribed the sensor device.

Below my updated code. I haven't done the operational hours yet. I have done enough javascript for today.

```
// Double rail with station and single rail at both ends
// For poweredup v6.4.1

const PoweredUP = require("node-poweredup");
const poweredUP = new PoweredUP.PoweredUP();

var DirectionEnum = {
  FORWARD: 1,
  BACKWARD: 2,
  
  properties: {
    1: {name: "forward"},
    2: {name: "backward"}
  },
  
	reverse: function(direction) {
		if (direction === DirectionEnum.FORWARD) {
			return DirectionEnum.BACKWARD;
		}
		return DirectionEnum.FORWARD;
	}
};

/**
 * @typedef Color
 * @property {number} BLACK 0
 * @property {number} PINK 1
 * @property {number} PURPLE 2
 * @property {number} BLUE 3
 * @property {number} LIGHT_BLUE 4
 * @property {number} CYAN 5
 * @property {number} GREEN 6
 * @property {number} YELLOW 7
 * @property {number} ORANGE 8
 * @property {number} RED 9
 * @property {number} WHITE 10
 * @property {number} NONE 255
 */

var ColorSpeedEnum = {
	BLACK: {name: "Black", colorId: 0,  speed: 0, sleepMilis: 0, reverse: false},
	PINK: {name: "Pink", colorId: 1,  speed: 0, sleepMilis: 0, reverse: false},
	PURPLE: {name: "Purple", colorId: 2,  speed: 0, sleepMilis: 0, reverse: false},
	BLUE: {name: "Blue", colorId: 3,  speed: 0, sleepMilis: 0, reverse: false},
	LIGHT_BLUE: {name: "Light blue", colorId: 4,  speed: 0, sleepMilis: 0, reverse: false},
	CYAN: {name: "Cyan", colorId: 5, speed: 0, sleepMilis: 0, reverse: false},
	GREEN: {name: "Green", colorId: 6, speed: 0, sleepMilis: 5000, reverse: false},
	YELLOW: {name: "Yellow", colorId: 7,  speed: 0, sleepMilis: 0, reverse: false},
	ORANGE: {name: "Orange", colorId: 8,  speed: 0, sleepMilis: 0, reverse: false},
	RED: {name: "Red", colorId: 9, speed: 0, sleepMilis: 6000, reverse: true},
	WHITE: {name: "White", colorId: 10,  speed: 0, sleepMilis: 6000, reverse: false},
	
	lookupByColorId: function(colorId) {
		//log('got colorId: ' + colorId);
		for (var app in ColorSpeedEnum) {
			if (isNaN(Number(app))) {
				if (ColorSpeedEnum.hasOwnProperty(app) && ColorSpeedEnum[app].colorId === colorId) {
					//log('got color: ' + ColorSpeedEnum[app].name);
					return ColorSpeedEnum[app];
				}
			}
		}
	 }
};

[... 224 lines cut ...]
}

function getFormattedDate(timeStamp) {
    var date = timeStamp === undefined ? new Date() : new Date(timeStamp);

    var month = date.getMonth() + 1;
    var day = date.getDate();
    var hour = date.getHours();
    var min = date.getMinutes();
    var sec = date.getSeconds();

    month = (month < 10 ? "0" : "") + month;
    day = (day < 10 ? "0" : "") + day;
    hour = (hour < 10 ? "0" : "") + hour;
    min = (min < 10 ? "0" : "") + min;
    sec = (sec < 10 ? "0" : "") + sec;

    return date.getFullYear() + "-" + month + "-" + day + " " +  hour + ":" + min + ":" + sec;
}

function isOperationHours() {
	var date = new Date();
	return date.getHours() >= 8 && date.getHours() < 20;
}
```

cpseager, 2020-06-06:
You could put the sensor into LED light mode (Mode.LED)  and then switch all the LEDs off via

setColor(0)

That would save most of the power I think. Then switch back to 

Mode.COLOR_AND_DISTANCE

when needed.

RaymondNg2, 2020-06-07:
Thanks @cpseager. I did not know it was possible to turn off the led. I added it to my script and it works fine. Because my original problem has been addresses I will close this issue.

closed 2020-06-07 by RaymondNg2