nathankellenicki/node-poweredup #92: Set RGB colors to device RGBLight in hub TechnicMediumHub

tthiery, 2020-06-06:
Is it a known issue that setRGB is not working on TechnicMediumHub internal RgbLight (port 50)?

I modified the WebBluetooth sample with the following adjustment

````javascript
                //led.setColor(color); // Set the color
                led.setRGB(0x66, 0x33, 0x99);
````

Resulting behavior: Light stays blue.

I am just looking for a confirmation that this is a known issue.

Full disclosure: Working on a .NET library and I am also not able to get SetRgbColors running but only SetRgbColorNo. Both return a feedback 0x0A (that is busy/full) but ColorNo actually work. I thought I did not understand the protocol yet, but I see the same behavior also here.

ps: Thanks for this library. It is a huge inspiration and very helpful.
pps: Is there a reddit / secret community for library implementers?

aileo, 2020-06-06:
I tried to reproduce but it works just fine for me with the following code:

```js
let color = 1;
setInterval(() => {
    const colors = [
        [255, 0, 0],
        [255, 255, 0],
        [0, 255, 0],
        [0, 255, 255],
        [0, 0, 255],
        [255, 0, 255],
        [0x66, 0x33, 0x99],
    ];

    const hubs = poweredUP.getHubs(); // Get an array of all connected hubs
    hubs.forEach(async (hub) => {
        const led = await hub.waitForDeviceByType(PoweredUP.Consts.DeviceType.HUB_LED);
        led.setRGB(...colors[color]);
    });
    color = (color + 1) % colors.length;

}, 2000);
```

You can enable the debug output by running `localStorage.debug = '*'` in the browser console and refreshing the page, maybe it will help to understand the problem.

tthiery, 2020-06-06:
I figured what made your sample work and not my library. The `node-poweredup` package does a *Port Input Format Setup Single Message* (`this.subscribe`) **before** the invocation of the *DirectWriteModeData* (`this.send`). Formatting that port port/mode it will later write does the trick (it does not matter, whether a interval or notification is set, as long as the port/mode is formatted).

There is some issue in the WebBluetooth sample. The invocation to setRGB/setColor does not return in some period of times from (like 5 minutes till e.g. an unrelated code change happened) the await (it does not write a byte array log entry also). I saw that several times, independent of this issue.

Watch for the `this.subscribe` call. it has ugly defaults of delta interval 1 and enabled updates. Can burn battery in cases.

From my side we can close this.

nathankellenicki, 2020-06-12:
@tthiery Thanks for noticing that. I must admit the subscribe command this code uses has lingered since the first implementation, long before the official LWP documentation came out, so we weren't sure what the individual bytes did.

I'll take a note to make this a little more performant.

Closing this issue as the original issue is solved. :)

closed 2020-06-12 by nathankellenicki

tthiery, 2020-06-12:
Thanks @nathankellenicki for your work on this project.