nathankellenicki/node-poweredup #93: writeValueWithoutResponse for better performance

dlech, 2020-06-06:
FYI, I recently implemented a new feature in Chromium 85 to specifically allow writing a BLE GATT characteristic without response. It would be cool if someone could try it and see if it makes a difference with this library.

Since the LEGO LWP3 characteristic says that it supports both write with response and write without response, Windows, Mac and Linux currently default to writing with response in Chrome, which is much slower than writing without response. With the new APIs in Chrome, it is possible to select which type of write to use. So if you are continuously sending commands to a Powered Up hub, you should be able to see a difference when comparing writing with response and without.

Summary: https://chromestatus.com/feature/5152041850634240

Get Chrome 85: https://www.google.com/chrome/dev/

nathankellenicki, 2020-06-12:
Awesome work @dlech! I was wondering if the Chromium team would ever implement this - I noticed poor performance when sending multiple commands to the hub in quick succession and assumed it was down to the usage of this.

Easier to implement it yourself it turns out. ;)

Chrome public is still on 83 for me. When 85 is released to the wild I'll migrate this library over to using these commands. Thanks for the heads up!

nathankellenicki, 2020-12-23:
@dlech I've made this improvement in `v6.8.1` of node-poweredup, thanks for committing it to Chromium in the first place!

closed 2020-12-23 by nathankellenicki