nathankellenicki/node-poweredup #95: How to find port number for speaker & speedometer?

matmar10, 2020-06-20:
First off: AMAZING LIBRARY. I cannot express in words how giddy (literally: giggling like a child) I was when I was able to connect to and control my son's duplo train within 2 minutes of discovering this library. This stuff is like PURE MAGIC! Amazing work.

Question: only the following port numbers are documented:
```
export const PortMap: {[portName: string]: number} = {
    "MOTOR": 0,
    "COLOR": 18,
    "SPEEDOMETER": 19
};
```

The docs mention the following devices for Duplo train base:

> In addition, the Hubs themselves have certain built-in features which this library exposes.
> - RGB LED/Headlights
> - Speaker
> - Speedometer
> - Motor
> - Color and Distance Sensor
> - Button

How can I determine the port of the other devices? Trial/error by incrementing the port and manually creating the device? If this is the right approach, I'll do this and open a PR which includes the discovered port numbers.

Or is there a simpler approach I'm missing (e.g. ports are already listed in code someplace I didn't see?)

Thanks again for this amazing library. Hours of fun ahead with this!

matmar10, 2020-06-20:
Got it! Looks like:
```
await hub.waitForDeviceByType(PoweredUP.Consts.DeviceType.DUPLO_TRAIN_BASE_SPEAKER);
```
Seems like the way to go.

I'll be raising a PR soon that maps the built-in tones to constants.

closed 2020-06-20 by matmar10

nathankellenicki, 2020-06-20:
Glad you found it! It's a good point - the names were more appropriate in an older version of this library where everything was named, but not so much anymore now that each device is its own class. I should either device to name everything or make it so names arent required (which they are, especially for hub types with removable devices).

For reference, known Duplo train speaker consts are here, however if I've missed any feel free to let me know! https://github.com/nathankellenicki/node-poweredup/blob/master/src/consts.ts#L167

matmar10, 2020-06-20:
Awesome, thanks! Yes I found those - I mistyped and meant tone (e.g. playTone). I've recorded those and described what they are to the best of my ability - they're all based on a major scale. 

Question about `RGB LED/Headlights` - where can I mess about with that? i didn't see a specific device. Is there something generic?

FYI I plan to write up a full tutorial / examples for other Duplo train owners. 

This stuff is just too cool and I am geeking out now, haha.

nathankellenicki, 2020-06-20:
Glad you're enjoying it - it is a lot of fun! :D Especially when you realise that all the new Powered Up/Control+/Mindstorms range speaks the same protocol and a whole world of possibilities opens up.

Regarding the headlights, it's emitted as a standard HUB_LED object, since that's actually what it is - just with a plastic light channel leading up to the front of the locomotive. https://github.com/nathankellenicki/node-poweredup/blob/master/src/consts.ts#L61

Super cool about the tone! I hadn't spent much time figuring it out when I played with it, so a PR is welcome with any findings you have! :)

Similarly for an example or two to place in the examples directory - while I do own a couple and obviously implemented them, I kind of neglect them a bit and forgot to write an example. :)