nathankellenicki/node-poweredup #98: Help:  hub.waitForDeviceAtPort is not a function

mylogon341, 2020-06-23:
When trying to mimic the train example, I am getting the following error 
```
0|index    | TypeError: hub.waitForDeviceAtPort is not a function
0|index    |     at PoweredUP.poweredUP.on (/home/pi/train/index.js:45:30)
```
my code looks like this
```
poweredUP.on("discover", async (hub) => { // Wait to discover a Hub
    await hub.connect(); // Connect to the Hub

    console.log("connected")
    const motor = await hub.waitForDeviceAtPort("A")
```
I am getting around this for motor speed by using 
```
    hub.setMotorSpeed("A", `${speed}`)
```
which works fine, but not much use if I want to use 
```
    motor.brake()
```
as I cant get an instance of the motor.
Any help appreciated.
Luke

aileo, 2020-06-23:
I dont have the real solution but if you are sure the motor is already connected on port A then you can use `hub.getDeviceAtPort("A")` instead.

nathankellenicki, 2020-06-23:
@mylogon341 The hint for me is that `setMotorSpeed` works for you - this function was removed in 6.0.0 in the library. Similarly, `waitForDeviceAtPort` is a newer method introduced in 6.0.0.

You need to upgrade to the latest version. :)

mylogon341, 2020-06-23:
5.0.4 🤦 
That one always trips me up when I come back to something after a while.
Sorted. Thanks

closed 2020-06-23 by mylogon341