nathankellenicki/node-poweredup #103: New Mario interactive figure (71360) is a poweredup device

nutki, 2020-08-01:
Here are the details I have found so far:
Device Id: 67
Port 0: type 0x47 - 3 axis (?) accelerometer
Port 1: type 0x49 - color (and barcode?) sensor
Port 2: type 0x4a - 6bit pants detector (the front bottom of the figure has 6 binary switches to sense different power up accessories)
Port 3: type 0x46 - ??? (modes: COUNTERS 16 bytes, ASSETS 2x32bit, GEVER 4x32bit) 
Port 6: type 0x14 - standard voltage port
All modes are input only so if there are ways to control the screen and sound then those must use some non standard means.

nutki, 2020-08-01:
After the initial update from version 1.1 as shipped to 2.0 there are some changes:
- accelerometer port lost most modes (only GEST and RAW left)
- port 2 completely changed to now have modes: CHAL (2x16bit), VERS (4x32bit; maybe the same as GEVER in 1.1), EVENTS (2x16bit), and DEBUG (4x32bit)
Now I wish I investigated more the port 2 modes before upgrading. Either way port 2 seems to be the interface used by the mobile app now (as it does not seem to drive the game but only serves as a secondary display for it and probably this is how it obtains its state like the current score).

nathankellenicki, 2020-08-02:
Thanks @nutki. :) I'm still waiting for mine arriving - I moved house recently, and I forgot to change the address. Who knows when it'll arrive.

Have you seen this repo? https://github.com/bricklife/LEGO-Mario-Reveng

I think together there's some nice info out there - I'll add support to this library when mine arrives. Thanks for the summary!

nutki, 2020-08-02:
No I have not seen this. Thanks for the link. I tried to search for it but it was hard using a generic keywords.

nathankellenicki, 2021-04-06:
Old topic, but I added support for the Mario hub and sensors a while back! So closing. Thanks again!

closed 2021-04-06 by nathankellenicki