nathankellenicki/node-poweredup #104: [Question] spike hub to wedo 2.0

radu022003, 2020-08-11:
hi, i have successfully connected over BLE a Lego Spike Hub and Lego WeDo 2.0.
Next step was to exchange some data between them, so i tried to change the WeDo 2.0 LED color.
Following your source code i could see that for setting the LED, 2 BLE Characteristics need to be used.
The question is, how to do this using the micropython on Spike hub? Since i saw that the send function of the connection object does not support the second parameter, the BLE Characteristic.

nathankellenicki, 2020-08-11:
Hi @radu022003, I'm afraid I don't know the answer to your question, as I haven't played around with Python on the SPIKE Prime hub in a while (this repo is for Node.js implementation for Powered Up devices).

@nutki has done more work on this recently over at https://github.com/nutki/spike-tools, perhaps he'll know better than me.

nathankellenicki, 2020-08-11:
Closing as unrelated.

closed 2020-08-11 by nathankellenicki